PiSCSI/piscsi #792: Slow and unstable operation on Mac Plus

rdmark, 2022-08-18:
# Info

- Which version of Pi are you using: 3B+, Zero
- Which github revision of software: any (tested the latest develop, as well as the 5 most recent stable releases, as far back as Oct 2021)
- Which board version: any
- Which computer is the RaSCSI connected to: Mac Plus (tested on two otherwise working Pluses)

# Describe the issue

While using RaSCSI as the boot drive for a Mac Plus, I experience very slow boot times, and once booted the system software will almost immediately throw some kind of bus or address error and crash.

If you boot off a floppy and try to format a blank RaSCSI disk image, HD SC Setup always crashes on the mounting volume step.

rdmark, 2022-08-18:
Without any trace logging enabled, there was one error raised during operation. It occurred when changing system folder for booting. 

```
$ sudo ./rascsi -ID0 macpack.hda
SCSI Target Emulator RaSCSI Reloaded version 22.08 --DEVELOPMENT BUILD-- (Aug 18 2022, 09:51:30)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2022 Contributors to the RaSCSI Reloaded project
Connect type : FULLSPEC
[2022-08-18 10:22:42.350] [info] Set log level to 'info'
[2022-08-18 10:22:42.355] [info] Validating: operation=ATTACH, device id=0, unit=0, type=UNDEFINED, device params='interfaces=macpack.hda':'file=macpack.hda', vendor='', product='', revision='', block size=0
[2022-08-18 10:22:42.357] [info] Executing: operation=ATTACH, device id=0, unit=0, type=UNDEFINED, device params='interfaces=macpack.hda':'file=macpack.hda', vendor='', product='', revision='', block size=0
[2022-08-18 10:22:42.358] [info] Attached SCHD device, ID 0, unit 0
[2022-08-18 10:22:42.358] [info] +----+-----+------+-------------------------------------
[2022-08-18 10:22:42.358] [info] | ID | LUN | TYPE | IMAGE FILE
[2022-08-18 10:22:42.358] [info] +----+-----+------+-------------------------------------
[2022-08-18 10:22:42.358] [info] |  0 |   0 | SCHD | /home/dmark/images/macpack.hda
[2022-08-18 10:22:42.358] [info] +----+-----+------+-------------------------------------
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  0 |   0 | SCHD | /home/dmark/images/macpack.hda
+----+-----+------+-------------------------------------
[2022-08-18 10:26:52.620] [error] virtual void SCSIDEV::Receive() Not able to receive 512 bytes of data, only received 163. Going to error
```

uweseimet, 2022-08-18:
@rdmark This is a duplicate of https://github.com/akuker/RASCSI/issues/656, isn't it?

rdmark, 2022-08-18:
@uweseimet It does read very similarly. Thanks for reminding me about this one. Let me see if the silverlining workaround does it for me too.

rdmark, 2022-08-19:
Confirmed that this is a duplicate with #656 . Closing.

closed 2022-08-19 by rdmark