PiSCSI/piscsi #799: VAXstation 3100 Issues with booting

Pacjunk, 2022-08-21:
# Info

- Which version of Pi are you using: Zero W
- Which github revision of software: Release v22.05.02 or later
- Which board version: FullSpec
- Which computer is the RaSCSI connected to: VAXstation 3100

Note this is a continuation from and summarisation of  #769

# Background

The VAXstation 3100 uses ROM code to boot from SCSI devices. This ROM "driver" is used until the operating system driver is loaded at which point it takes over. There are known limitations with this ROM code, the main one being the use of 6 byte SCSI commands only which limits usable boot disk sizes.

At boot time, the system issues an inquiry command, to gather basic information.
If the inquiry returns a SCSI response code of:

1 : The server sends a modesense6 command, requesting page $3F and length $0C
2 : The server sends a modesense6 command, requesting page $00 and length $0C

RASCSI returns SCSI level 2 and response level 2, so the second option occurs.
The 12 bytes requested is enough to hold the mode parameter block, but no page code data.

# Issue 1 - Page code 0 support

Up until release v22.05.02, page code 0 was accepted, the 12 bytes were returned and everything was happy.
After this release and, I believe, #699 page code 0 is now rejected, so the modesense6 command generates an error and the system will not boot. RASCSI will still function if the system is booted from a physical hard drive as the OS driver does not request page 0. The physical SCSI-2 hard drives that I have, return a SCSI response code of 2 but work correctly, so they **must** be responding to page code 0 - I believe this page is vendor specific.

# Solution 1

As the page code 0 support cannot be reinstated, the only other option is to change the reported SCSI response code (and probably the level) back to 1. This means the system will issue a request for page code $3F which is supported.
In the short term, I have modified my local copy of the code to report response code 1.
In the longer term, RASCSI will probably have to provide support for emulating SCSI-1 devices by modifying the response to the inquiry command based on other settings/filenames/etc.

# Issue 2 - Checking requested and returned length

Around the same time frame, the code was also changed to check the length of the modesense request against the actual data generated and produce an error message in case of mismatch. Page code $3F means to return **all** code pages and is a lot longer than the 12 byte mode parameter block that is requested. This caused another error to occur.

# Solution 2

The RASCSI code has already been modified to truncate any data generated by the modesense command to the length requested.

# Issue 3 - Use of return codes of 0 to flag an error

Solution 2 has revealed another issue. The function ModePageDevice::AddModePages has two exit points. If there are no pages returned due to an invalid code (such as 0!), then the function exits with a code of '0' - signalling an error, otherwise the function exits returning the number of bytes generated (and truncated by the solution to issue 2).

The problem with the VAXstation requesting only 12 bytes (and that being provided by the mode parameter block), is that the requested number of bytes from the page codes is effectively zero. When the function exits with a valid value of zero (not in error), the calling procedure mistakes this for an error condition.

# Solution 3

The quick and dirty solution is to change the error flag to something else. I have used -1 as a quick fix. A bigger issue is that this error flag mechanism is used throughout the RASCSI code and has the potential to cause similar issues.
In the long term, there may need to be a better way of handling exceptions that avoids these issues.

uweseimet, 2022-08-22:
@Pacjunk The latest sasi_removal branch may fix issues 2 and 3. Can you please test? Ensure that you have a backup of your data. I have not yet found problems with my environment, but there were many changes and the code has not yet extensively been tested.
If there is still a mode page problem, can you try to identify which part of the code causes it? The size return from the callee is now always supposed to be a valid value and does not indicate an error condition anymore.

Pacjunk, 2022-08-24:
I did a rebuild, but can't get the web UI to start now. Is this expected? I assume this is more for @rdmark 

```
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]: Serving rascsi-web...
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]: 2022-08-24 11:37:42,084 ERROR app.py:1455 Exception on / [GET]
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]: Traceback (most recent call last):
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "/home/pi/RASCSI/python/web/venv/lib/python3.7/site-packages/flask/app.py", line 2070, in wsgi_app
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     response = self.full_dispatch_request()
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "/home/pi/RASCSI/python/web/venv/lib/python3.7/site-packages/flask/app.py", line 1515, in full_dispatch_request
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     rv = self.handle_user_exception(e)
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "/home/pi/RASCSI/python/web/venv/lib/python3.7/site-packages/flask/app.py", line 1513, in full_dispatch_request
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     rv = self.dispatch_request()
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "/home/pi/RASCSI/python/web/venv/lib/python3.7/site-packages/flask/app.py", line 1499, in dispatch_request
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     return self.ensure_sync(self.view_functions[rule.endpoint])(**req.view_args)
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "web.py", line 112, in index
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     server_info = ractl_cmd.get_server_info()
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "/home/pi/RASCSI/python/common/src/rascsi/ractl_cmds.py", line 57, in get_server_info
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     if mappings[dtype] == proto.PbDeviceType.SAHD:
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:   File "/home/pi/RASCSI/python/web/venv/lib/python3.7/site-packages/google/protobuf/internal/enum_type_wrapper.py", line 117, in __getattr__
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]:     self._enum_type.name, name))
Aug 24 11:37:42 raspberrypi RASCSIWEB[678]: AttributeError: Enum PbDeviceType has no value defined for name 'SAHD'

```

rdmark, 2022-08-24:
@Pacjunk I pushed a commit to the same branch that should remove all support for SASI in the Web UI.

Pacjunk, 2022-08-24:
Thanks, I'll do a rebuild...

Pacjunk, 2022-08-24:
@rdmark Yes, that fixed it. Thanks
@uweseimet My system boots and appears to work fine. So far, so good from my perspective.

uweseimet, 2022-08-24:
@rdmark What about the SCSI level? I guess your system still requires SCSI-1 instead of SCSI-2? If it does, please confirm that with the unmodified sources from the sasi_removal branch the only change you need is

```
vector<BYTE> SCSIHD::InquiryInternal() const
{
	return HandleInquiry(device_type::DIRECT_ACCESS, scsi_level::SCSI_1, IsRemovable());
}

```
instead of
```
vector<BYTE> SCSIHD::InquiryInternal() const
{
	return HandleInquiry(device_type::DIRECT_ACCESS, scsi_level::SCSI_2, IsRemovable());
}

``` 

From a source code perspective this should be the only change required to make a drive SCSI-1 instead of SCSI-2.

Pacjunk, 2022-08-24:
Yes, I still need to hack the code to change the response code (I leave the SCSI level at 2).

I change the file **primary_device.cpp** at this line:

`buf[3] = level >= scsi_level::SCSI_2 ? scsi_level::SCSI_2 : scsi_level::SCSI_1_CCS;`

Changing where you suggest would also work and is probably easier. I was just looking for the absolute minimum change.
As I've mentioned above, I have physical HDDs that response with SCSI-2 that also work, but they must be doing something with page code 0.

uweseimet, 2022-08-24:
@Pacjunk Regarding page code 0 I wonder whether there is anything wrong with the analysis so far.
I do not understand why a computer would require page 0 for everydays work, i.e. for everything thati is not a very special hard disk-related operation. In your case, if only 12 bytes are requested and if these 12 bytes are returned everything works, correct? So what is the relevance of these 12 bytes? It sounds as if your computer is happy as long as anything is returned.
If I were interested in the page code header, but not in the page code data, I would use page code $3F, which will usually yield a result, and not page $00.

Pacjunk, 2022-08-25:
The relevance is that the 12 bytes contain the mode parameter block. Number of blocks on the device and block size. Apparently this is all the system needs to boot. Why the code requests $00 for SCSI-2 and $3F for SCSI-1 I do not know. Maybe a bug or maybe it is intentional. Whichever one it uses, it is happy with the information in those 12 bytes. I have a few different versions of the ROM and they all do the same thing.

uweseimet, 2022-08-25:
@Pacjunk Maybe READ CAPACITY was introduced with SCSI-2. At least with SCSI-2 this is the recommended method to inquire the number of blocks. For devices where this is not supported MODE SENSE can probably be used as a fallback. The header maybe more helpful than the data from the geometry page in this case, because the geometry information may help with calculating the capacity by multiplying some values, but it does not necessarily reflect the real geometry, so that such a calculation might lead to a wrong result.
But not using $3F as a safer bet than $00 is strange.

Pacjunk, 2022-08-25:
It is what it is!  Just a quirk of that system, and not something that I'm capable of changing. At least there is a workaround (using SCSI-1). The ROMs are dated from 1989 to 1991 before the spec was finalised, so maybe it behaves according to an older draft.

uweseimet, 2022-08-26:
@Pacjunk I guess I will add support for .hd1 (i.e. generic SCSI-1 drive) after the 22.08 release, with a separate PR. I don't know whether the rest of the code (e.g. bus phase handling) is compliant with SCSI-1, but at least in your case just dealing with the SCSI level and response format appears to be sufficient.

closed 2022-09-08 by uweseimet