PiSCSI/piscsi #809: Re-add SASI Functionality

akuker, 2022-08-26:
# Info

The team is working to remove SASI support from the current RaSCSI code. The current design/architecture makes it very difficult to make changes and has substantial maintainability impacts. 

This issue is to gauge interest in SASI support. 

# How can you help?

If you are interested in having SASI support re-added into RaSCSI, please leave a comment with some information so we can understand your use case. Some info that I would like to collect:
- What target hardware platform are you using (including add-on card/mod if applicable)?
- What software version are you using?
- What features/drive types do you typically use? (Hard disk, CD-ROM, Magneto Optical, etc)

Feel free to add duplicates to the comments chain. Or, if you use multiple systems, feel free to add them to the list

# How else can you help?

If you are in the continental US and have an extra SASI host that you are willing to part with for a reasonable price, feel free to reach out to me on [discord](https://discord.gg/PyS58u6).

rdmark, 2022-08-27:
@philpem @drdpj @wohali Tagging you for feedback, since you have raised your interest in SASI support in the past. It would be very helpful if you helped us understand your hardware and use cases here!

uweseimet, 2022-08-30:
If somebody wants to re-add SASI, this is how it should be done:

- The scsi_ctrl.cpp code from release 22.08.01 can be used for reference.
- The SASI controller class has to be a subclass of AbstractController.
- The device implementations (derived from PrimaryDevice) must not know anything about controller specifics. They only know AbstractController.
- The rascsi process (rascsi.cpp and related) must also not know anything about controller implementations, but only about AbstractController.

wohali, 2022-09-06:
Here are my direct use cases:

My understanding is that GIMONS' [original RaSCSI](https://github.com/RaSCSI/RaSCSI) was for the X68k. I understand he isn't involved in the project anymore (no need to rehash this). X68k is still relevant to me.

`sasidump` has been useful in rescuing old SASI drives, including those on e.g. an old Plessey PDP-11 SASI controller that is largely undocumented I have here. It would be nice to have an emulator that supported this controller.

Finally I have several Sony/Panasonic/Pinnacle Micro 650MB MO drives, which are SASI and sometimes lived behind a (fairly buggy) SASI-SCSI bridge. Quirks of these drives from oddball controllers mean they often don't emulate right purely on SCSI buses, and in one instance I have one connected directly to a SASI controller. Would be good to replace it, it's an increasingly unreliable mechanism.

I suspect others may have a need for SASI (ACSI) support for Atari machines.

uweseimet, 2022-09-06:
@wohali Regarding Ataris, what you need is SCSI support, for both ACSI and SCSI. I am using RaSCSI with a MegaSTE (internal host adapter), a TT and a Falcon.

hlide, 2022-11-12:
I'm owner of two X68000 PRO and they are SASI. I see the removal of SASI has a very bad move since you deprive us of the possibility to use your version of RaSCSI (which is open-source) for those X68000 with SASI.

akuker, 2022-11-12:
> I'm owner of two X68000 PRO and they are SASI.

Are you using this fork of the RaSCSI software? 

> I see the removal of SASI has a very bad move since you deprive us of the possibility to use your version of RaSCSI (which is open-source) for those X68000s with SASI (that is, models under XVI).

The previous release of RaSCSI still supports SASI, and is available for download. There are archives of GIMONS's original code available from multiple sources (including archive.org). 

None of the developers have access to a X68000 (yet). SASI functionality was broken for several releases and no one noticed because we don't have a way to test it. 

The plan is to re-add this capability in the future. If you have a X68000 that is available for sale at a low price, or if you are willing to volunteer time to test/troubleshoot/support the SASI functionality, it would help us get this functionality re-added sooner.

Please keep in mind that we are all volunteers and doing our best. Please take this ticket as proof that we have not permanently removed this functionality and have intentions of adding it back in.

rdmark, 2022-11-12:
> `sasidump` has been useful in rescuing old SASI drives, including those on e.g. an old Plessey PDP-11 SASI controller that is largely undocumented I have here. It would be nice to have an emulator that supported this controller.

@wohali When did you last use `sasidump` and did it work properly then? The reason for deprecating the code is that it hadn't been tested for years. If you can confirm that `sasidump` in the Aug 2022 release works properly, we can look into adding it back.

wohali, 2022-11-12:
It was prior to Aug 2022. I can attempt again in December - booked out the rest of this month, I'm afraid.

hlide, 2022-11-12:
> None of the developers have access to a X68000 (yet). SASI functionality was broken for several releases and no one noticed because we don't have a way to test it.

Fine, I understand the point as I wasn't aware of that issue. I was glad to see there is an attempt to recreate RaSCSI in a true open-source form. What I fear is "This issue is to gauge interest in SASI support" means "No SASI support" at the end. That occurs too often. SASI may sound like a niche because I think it mostly touch X68000 folks (a lot of them being Japanese) and I won't expect for them to advocate their concerns here, hence my reaction.

> The plan is to re-add this capability in the future. If you have a X68000 that is available for sale at a low price, or if you are willing to volunteer time to test/troubleshoot/support the SASI functionality, it would help us get this functionality re-added sooner.
- I have no X68000 available for sale but I can certainly test on my working X68000 PRO (SASI and maybe SCSI with an extension card). I also have a working X68000 Compact XVI (SCSI but the external connector is non standard).
- I have an RaSCSI + PI4 that I didn't prepare due to crazy time, crazy work, etc. because I think I miss some components: µSD and fitting SASI/SCSI cables to complete it. I also have an SCSI card (unknown state though) slotted in the PRO. If I have clear instructions of what I need to make it work with your software, I can certainly test it. My main obstacle is, I think, the cables I need to get as they are not really standard. 
- I'm a C++ developer on PC (mostly Windows, Linux marginally). While I coded on Arduino, I never coded for Raspberry.

akuker, 2023-09-23:
>  What I fear is "This issue is to gauge interest in SASI support" means "No SASI support" at the end. 

[image]

We may be slow, but we haven't forgotten the x68000 community! (See previous comment about volunteers :) )

Last week I purchased an original Sharp x68000 and am trying to "learn" it. (If I knew Japanese it might help)

So far, I'm able to boot the device using a Human 68k version 3.02 image. It seems that basic communication is working with the regular SCSI functionality in PiSCSI. I haven't been able to launch any games from the HD image, but that could be user error, or I just might not have enough memory. (This is a very basic, original machine - I think with only 1MB RAM).

This ticket hasn't died, it has just been a slow process for me to track down a x68000 to tinker with.

jjlopezg, 2024-02-14:
Hello, I have recently bought an x68000, I would like to be able to buy the pcb to assemble and carry out tests until it works correctly. Could you tell me where I can buy the PCB alone. I have pytho experience to help. thank you so much

jjlopezg, 2024-02-14:
Hello, I have recently bought an x68000, I would like to be able to buy the pcb to assemble and carry out tests until it works correctly. Could you tell me where I can buy the PCB alone. I have pytho experience to help. thank you

rdmark, 2024-02-15:
@jjlopezg Everything you need to know is written on the landing page of www.piscsi.com

You either buy a kit from one of the resellers, or produce your own PCBs via a PCB manufacturer.

gustavosarmento, 2026-08-24:
It's a Pi 2 Model B Rev 1.1.

rdmark, 2026-08-24:
@gustavosarmento got it, if I read the spec sheet correctly this is a Cortex-A7 32 bit CPU which maps to the `armhf` architecture in the Debian cross compilation framework. This is what I’ll build for you. 

Would it be possible for you to arrange a fresh microSD card with Raspbian Trixie for testing?

gustavosarmento, 2026-08-24:
Sure thing. I have a spare microSD assigned for this.

rdmark, 2026-08-24:
@gustavosarmento please try these

piscsi: https://drive.google.com/file/d/1QyduRwGfs_3NPhNVcEoUAnPFB8b7DlI3/view?usp=sharing
piscsi-web: https://drive.google.com/file/d/1kjtzh0GQS9HQE9zTHLK2Fof_S59AyI7h/view?usp=sharing

transfer to your RPi and install with:

```
sudo dpkg -i piscsi_26.8.1-1_armhf.deb piscsi-web_26.8.1-1_armhf.deb
```

if you are missing dependencies dpkg will tell you, so install them one by one with apt and you should eventually be able to install them

there's also a draft of an INSTALL.md here with a bunch of package you can install in one go: https://github.com/PiSCSI/piscsi/blob/c881412d4b27d63e5a3351d962b2526d394fa1b9/INSTALL.md

I'm curious to hear how it goes!

gustavosarmento, 2026-08-24:
Do I need to have a previous PiSCSI version installed? I'm struggling a bit with the installation...

rdmark, 2026-08-24:
@gustavosarmento please tell me the exact steps you took and the terminal output: these packages should be installed on a vanilla Raspbian Trixie system, no previous version of PiSCSI must be installed

gustavosarmento, 2026-08-24:
1) Installed Raspberry Pi OS Lite Trixie on microSD card (via RaspberryPi Imager);
2) ssh to Pi;
3) apt update, apt full-upgrade and reboot;
4) Copied both files to Pi via scp;
5) executed your above mentioned command;
6) Terminal output:

```
Selecting previously unselected package piscsi.
(Reading database ... 89956 files and directories currently installed.)
Preparing to unpack piscsi_26.8.1-1_armhf.deb ...
Unpacking piscsi (26.8.1-1) ...
Selecting previously unselected package piscsi-web.
Preparing to unpack piscsi-web_26.8.1-1_armhf.deb ...
Unpacking piscsi-web (26.8.1-1) ...
dpkg: dependency problems prevent configuration of piscsi:
 piscsi depends on libprotobuf32t64 (>= 3.21.12); however:
  Package libprotobuf32t64 is not installed.

dpkg: error processing package piscsi (--install):
 dependency problems - leaving unconfigured
dpkg: dependency problems prevent configuration of piscsi-web:
 piscsi-web depends on piscsi; however:
  Package piscsi is not configured yet.

dpkg: error processing package piscsi-web (--install):
 dependency problems - leaving unconfigured
Processing triggers for man-db (2.13.1-1) ...
Errors were encountered while processing:
 piscsi
 piscsi-web
```

gustavosarmento, 2026-08-24:
Oh, I did a **sudo apt-get install libprotobuf32t64** and apparently both PiSCSI and PiSCSI-web are running. But I can't get into the web interface.

rdmark, 2026-08-24:
@gustavosarmento that's good progress -- the web interface should be accessible over http on port 8080, e.g. http://192.168.0.123:8080

gustavosarmento, 2026-08-24:
Yay, that's it! Didn't know it was on port 8080.

Thanks a lot! Now, onto some testing...

gustavosarmento, 2026-08-24:
So far, I could boot .hds SCSI images on my x68000 PRO HD without issues. 

But I tried to upload a .hdf SASI image, and the web interface does not recognize this extension, thus I can't attach it.

rdmark, 2026-08-24:
@gustavosarmento aha, it is trivial to add support for .hdf files in piscsi-web, but it'll have to be tomorrow

as a workaround, try using scsictl on the terminal instead:

```
scsictl -i 5 -c a -t sahd -f yourfile.hdf
```

rdmark, 2026-08-24:
alternatively, rename the file with a .hds file ending, it should make no difference

gustavosarmento, 2026-08-24:
Nice! I managed to attach it, but it didn't boot the SASI image. Maybe I have to remember how to setup this on the x68k side, since I'm using just SCSI images recently.

Will resume the tests tomorrow. Thanks!

rdmark, 2026-08-25:
Is it possible to boot from a SCSI drive and also have a SASI drive connected? I assume they are separate interfaces and buses, not a single bus with two modes?

gustavosarmento, 2026-08-25:
I ejected the SCSI images, and attached only the SASI one. 

Thing is, I don't remember well the settings I have to change on the x68k side, to boot from SASI. As soon as I find my notes about it, I'll resume the testing. Probably later today.

gustavosarmento, 2026-08-25:
Well... no success so far.

I detached all SCSI images, attached only the SASI one, changed HD_MAX from 0 to 1 and upwards on Switch.x (this, as far as I remember, enables SASI boot), but no cigar. I can see some drive led activity, but it doesn't boot from the SASI image. Inserting a bootable floppy, I can't access the SASI image either.

Maybe there's still some bug left?

rdmark, 2026-08-25:
I took another look at the implementation: we are delivering a standards compliant SASI bus right now, however looking at the old RaSCSI code, and at what BlueSCSI v2 does, I see a handful of hard coded X68000 quirks -- let me port those over and pass you a new build

rdmark, 2026-08-25:
please try this: https://drive.google.com/file/d/1kz0ueojO6NwaadooId4X4Pj3KtL4uTsw/view?usp=sharing

you can use dpkg to install this on top of the previous version

piscsi-web is unchanged

gustavosarmento, 2026-08-25:
Updated, but still no progress.

A while ago, I installed an extended ROM on my x68k, so it won't be dependent on SRAM losing its settings every now and then. Maybe this is SCSI specific, and is getting on the way. I can disable it, but I'll have to open my x68k to do so. Will try to do it tomorrow.

That is, if you've ran out of ideas on the SASI implementation. :)

rdmark, 2026-08-27:
@gustavosarmento This points to an image/boot-record problem, not an extended-ROM or SASI transport problem.

The X68000 successfully reads:

- block 4 — byte offset `0x400`, where a SASI image’s partition table lives
- blocks 0–3 — byte offsets `0x000`–`0x3ff`, the disk IPL area

Every read completes with GOOD status. The “Fraudulent IPL” message occurs after the ROM has received those bytes and rejected the IPL content. For SASI, the boot ROM expects valid IPL code at the start of the image; a valid IPL normally begins with `0x60` (a 68000 `BRA` instruction). The partition table at `0x400` should normally begin with the `X68K` magic. [XEiJ SASI media layout](https://stdkmd.net/xeij/source/xeij-HDMedia.java.htm), [IPL validation detail](https://github.com/danifunker/rusty-backup/blob/main/docs/OPEN-WORK.md)

So I would rule out the extended ROM getting in the way. It is doing its expected job: loading the SASI IPL, then rejecting it.

Can you please run the following commands the PiSCSI host against the exact attached file:

```sh
xxd -g 1 -l 16 /var/lib/piscsi/images/SASI-40MB-MasterDiskV3.HDF
xxd -g 1 -s 0x400 -l 16 /var/lib/piscsi/images/SASI-40MB-MasterDiskV3.HDF
sha256sum /var/lib/piscsi/images/SASI-40MB-MasterDiskV3.HDF
```

Expected indicators:

```text
offset 0x000: starts with 60
offset 0x400: starts with 58 36 38 4b  # "X68K"
```

If either is absent, I would conclude that the image is corrupted, not a raw 256-byte-sector SASI image, or is not bootable for that machine. If both are present, we should compare the first 1024 bytes and SHA-256 with a known-working copy; the logs show PiSCSI is accessing precisely those bytes.

rdmark, 2026-08-27:
@gustavosarmento here are new deb builds, among other things we now map .hdf to the SASI device type, plus the sasidump tool has made a comeback

https://drive.google.com/file/d/1SQV0dAOKaUBZ-OK4hmIPzpGiGuCk45zx/view?usp=sharing, https://drive.google.com/file/d/1WKWbxlSrg52G2iEwiUqi52a1higAAm-1/view?usp=sharing

gustavosarmento, 2026-08-27:
Here we go!

offset 0x000: 60 00 00 ca 00 00 00 00 00 00 00 00 00 00 00 00  `...............
offset 0x400: 58 36 38 4b 00 02 79 30 00 02 79 30 00 02 ac c0  X68K..y0..y0....
sha256sum: 0f7275d0e15f387e761b9af2876a52bbcc137f12956f15042cf5d1b425b0d6d4

Next, installing the updates!

gustavosarmento, 2026-08-27:
Unfortunately, after updating, it insists on the IPL error.

rdmark, 2026-08-27:
yep, those bytes seem correct, although there might still be bad data elsewhere in 1k disk IPL

the greatest challenge right now is that we don't know if it's a piscsi bug or something wrong with the image or your setup -- do you have any other known good bootable images that you can try?

gustavosarmento, 2026-08-27:
I tried with an image a friend made me a while ago: same error.

I also tried both SASI images with a macOS x68k emulator, and both worked fine first time.

rdmark, 2026-08-28:
@gustavosarmento can you try running piscsi with a higher minimum command execution time, first 400 microseconds and then 800 microseconds if needed:

```sh
piscsi -s 400 …
```

```sh
piscsi -s 800 …
```

The default is 50 microseconds. This is a compatibility mode for the older initiators because it increases the settling time before the data and status phases.

rdmark, 2026-08-28:
@gustavosarmento also please use this new piscsi build, it is required for the -s argument to work for SASI devices.

https://drive.google.com/file/d/1JHp_07y-_l3zjQ3Tx-RRfBYhB_QenLCb/view?usp=sharing

gustavosarmento, 2026-08-28:
@rdmark 400 or 800 made no difference. But I got the following output from PiSCSI when executing either parameter. Have a look at the warnings, that appeared when I turned on the x68k:

pi@piscsi:~ $ sudo piscsi -s 800
SCSI Target Emulator PiSCSI (Backend Service)
Version 26.08.01
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2026 Contributors to the PiSCSI project
Connection type: FULLSPEC
[2026-08-28 12:22:20.248] [info] Set log level to 'info'
[2026-08-28 12:22:20.249] [info] Detected Raspberry Pi 2/3
[2026-08-28 12:22:20.251] [info] SCSI command execution time set to 800 microseconds
[2026-08-28 12:22:20.251] [info] Cleared reserved ID(s)
[2026-08-28 12:22:20.252] [info] No devices currently attached.
No devices currently attached.
[2026-08-28 12:22:28.909] [info] Detached all devices
[2026-08-28 12:22:28.915] [info] Reserved ID(s) set to 7
[2026-08-28 12:22:28.920] [info] Validating: operation=ATTACH, command params='locale=en', 'token=???', device=0:0, type=SAHD, device params='file=/var/lib/piscsi/images/SASI-40MB-MasterDiskV3.HDF', vendor='', product='', revision='', block size=256
[2026-08-28 12:22:28.921] [info] Executing: operation=ATTACH, command params='locale=en', 'token=???', device=0:0, type=SAHD, device params='file=/var/lib/piscsi/images/SASI-40MB-MasterDiskV3.HDF', vendor='', product='', revision='', block size=256
[2026-08-28 12:22:28.922] [info] Attached SAHD 0:0
[2026-08-28 12:22:29.959] [warning] (ID 0) - Received an unexpected command ($c2)
[2026-08-28 12:22:29.960] [warning] (ID 0) - Unexpected Data Out phase for command $c2
[2026-08-28 12:22:51.112] [warning] (ID 0) - Received an unexpected command ($c2)
[2026-08-28 12:22:51.112] [warning] (ID 0) - Unexpected Data Out phase for command $c2
[2026-08-28 12:22:51.115] [warning] (ID 0) - Received an unexpected command ($c2)
[2026-08-28 12:22:51.115] [warning] (ID 0) - Unexpected Data Out phase for command $c2

rdmark, 2026-08-28:
@gustavosarmento the warnings here are harmless I believe, just normal no-op for unexpected commands for the phase

next up, a build where PiSCSI SASI now matches pre-fork RaSCSI’s default timing behavior:

- Command → STATUS: 100 µs
- Command → DATA IN/OUT: 200 µs
- Completed DATA transfer → STATUS: 20 µs

https://drive.google.com/file/d/1ozuls0ehyVc3qCaxzLQoJ8aco-i1cS5w/view?usp=sharing

new debug logging has been added as well, which we can leverage if it still doesn't work

gustavosarmento, 2026-08-28:
@rdmark Still didn't work. BUT...

I changed to my RaSCSI card installation, and guess what: SASI Boot isn't working with it either.

Now I'm more convinced that the extended ROM is getting on the way somehow. I never tried SASI boot after installing it, because the extended ROM itself made this unnecessary (since it bypasses the need for SxSI be installed on the S-RAM, I don't have the need to boot from SASI or disk to restore it, in case it erases/corrupts).

I'll open my x68k and disable it (there's a tiny jumper for it, so no need to remove it physically). But I'll do this during the weekend, since it's a bit of a time consuming job.

The question is... if this makes it work, we'll have to rollback some versions/to the first version, right?

rdmark, 2026-08-29:
@gustavosarmento if it works after disabling the extended ROM, I would try lowering the DATA IN/OUT/STATUS waits again -- having the same waits for SASI and SCSI is preferable

we can also test if it works without the X68000 quirks for a more "standards compliant" SASI, but they are proven in RaSCSI and BlueSCSI so I expect we need them regardless

gustavosarmento, 2026-08-29:
Now we got some progress! It at least tries to boot now, with the extended ROM disabled. It still gives an error (with both SASI images I have. The only difference is that one errors in Japanese, and the other in English). [Here's a link to a picture with the error](https://i.postimg.cc/DynMhPP6/x68k-SASI-error.jpg).

Maybe now it's some fine-tuning issue... or not. :D

rdmark, 2026-08-29:
@gustavosarmento agreed, this seems like real progress: This error occurs after a successful IPL handoff and is more diagnostic of the SxSI device-registration path than the earlier SASI boot transfer. We can pause the timing tests for now.

This confirms the SASI image is now booting far enough to start Human68k configuration. Please capture PiSCSI’s trace log from the moment SCSIHD.SYS begins loading, including every CDB and status afterward.  

Also send the CONFIG.SYS lines that load SASIIOCS.SYS and SCSIHD.SYS, especially the ID= options. The disk is now attempting to register a SCSI device at ID 0 while PiSCSI is configured as an SAHD SASI target at ID 0.  

We should test the two configurations separately:

- Native SASI: SAHD, 256-byte blocks, and a native SASI system disk without relying on SCSIHD.SYS for its boot disk.
- SxSI-over-SASI: a SCSI hard disk (SCHD, normally 512-byte blocks) at the configured SCSI ID, bootstrapped through the SxSI SRAM loader / MasterDisk workflow.

rdmark, 2026-08-29:
BTW, don't hesitate to share screen grabs that contain Japanese text: the original language is often more concrete than the translations, and I happen to be fluent in Japanese.

gustavosarmento, 2026-08-29:
Complete boot trace:

[trace.txt](https://github.com/user-attachments/files/31598419/trace.txt)

gustavosarmento, 2026-08-29:
[And the Config.sys file](https://i.postimg.cc/Zn6YNtfD/x68k-configsys.png)...

rdmark, 2026-08-29:
@gustavosarmento my hypothesis is that using the SCSIHD.SYS driver to boot a SASI drive leads to SCSI commands being sent to the SASI controller, which leads to the error state here:

from the trace logs:

At the registration failure, the X68000 sends a SCSI command:

```
INQUIRY  CDB $120000002000
```

That requests 0x20 = 32 bytes. PiSCSI’s SAHD responds with only 2 bytes which is correct for SASI but confuses a SCSI driver, then enters STATUS:

```
Data In phase
Sending data … length: 2
```

The X68000 then waits about 1.4 seconds before accepting status and its next apparent CDB is garbage ($002525252525), leading to the “Not initialized” result.

Comparing with the historical RaSCSI implementation:

- RaSCSI’s SASI controller does not dispatch 0x12 INQUIRY; its SCSI controller does.
- RaSCSI explicitly rejects SASI/SCSI mixing.

From your CONFIG.SYS:

```
DEVICE = \SYS\SASIIOCS.SYS
DEVICE = \SXSI\SCSIHD.SYS ID=0 X68SCSI.BRP IBMSTD.BRP 98SCSIHD.BRP
```

My interpretation (X68k experts can correct me):

- SASIIOCS.SYS adapts the X68000’s SASI host port for SCSI access.
- SCSIHD.SYS ID=0 then expects a SCSI disk target at ID 0.

Do you by any chance have other drivers and configuration examples that you can try?

gustavosarmento, 2026-08-29:
Actually, I'm a bit of a newcomer to the x68k family of computers... got my first one only 3 months ago. :/

So, I'm not that knowledgeable about it. But, if anyone has something else that I can try to test, I'll be glad to help!

gustavosarmento, 2026-08-29:
Oh, and as an additional info: now, without the extended ROM, I again tested with the RaSCSI firmware, and it worked first time. 

So, I believe it's indeed something on PiSCSI now.

rdmark, 2026-08-29:
@gustavosarmento can you please tell me the precise RaSCSI software version you're using? is the original by GIMONS, or an earlier version of this fork by akuker which subsequently became PiSCSI?

gustavosarmento, 2026-08-29:
```
SCSI Target Emulator RaSCSI(*^..^*) version 1.52(Feb 13 2022, 22:13:53)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2021 GIMONS
Connect type : FULLSPEC
```

gustavosarmento, 2026-08-29:
It's the latest GIMONS original, as far as I know.

gustavosarmento, 2026-08-29:
And [just in case the hardware version matters](https://i.postimg.cc/y6hTmfJx/x68k-Ra-SCSI.jpg)...

rdmark, 2026-08-29:
can you please share details how you launch RaSCSI? alternatively, the commands you use to attach the various devices etc.

I'm trying to grasp what black magic RaSCSI use to handle the SxSI driver, since SCSI-over-SASI is not quite standards compliant, the mixing of SASI and SCSI commands is tripping up PiSCSI

and as a side note, don't hesitate to attach images directly here in the GitHub comments -- I'm struggling to access the ones you share via postimg.cc

gustavosarmento, 2026-08-29:
I followed [this guide](https://www.mushagaeshi.com/2018/02/24/howto-setup-rascsi-scsi-device-emulator/) to setup my RaSCSI (well, at least as good as the translation goes).

My startup script is as follows:

```
[Unit]
Description=RaSCSI
After=syslog.target

[Service]
Type=simple
WorkingDirectory=/opt/rascsi
#SASI boot
ExecStart=/usr/bin/sudo fullspec/rascsi -ID0 /var/rascsi/SASI-40MB-MasterDiskV3.HDF
#SCSI boot
#ExecStart=/usr/bin/sudo fullspec/rascsi -ID0 /var/rascsi/SxSI-SCSI_v302_disk1.hds -ID1 /var/rascsi/SxSI-SCSI_v302_disk2.hds -ID2 /var/rascsi/SxSI-SCSI_v302_disk3.hds -ID3 /var/rascsi/SxSI-SCSI_v302_disk4.hds -ID4 BRIDGE
TimeoutStopSec=5
StandardOutput=null

[Install]
WantedBy = multi-user.target
```

rdmark, 2026-08-29:
minor remark: in old school RaSCSI you're supposed to use the `-HD0` argument for .hfd images in SASI mode, not `-ID0` which is reserved for SCSI devices

anyhow, I see two ways forward:

option one, use the stock SASI driver instead of SxSI when booting from a SASI device; very simple CONFIG.SYS configuration example inspired by discussion in https://spend-carefree.hatenablog.com/entry/2019/08/27/001629 (Japanese):

```
DEVICE = \SYS\SASIIOCS.SYS
```

option two, recompile the GIMONS rascsi binary with trace logging support so that we can see what's going on on the bus when booting the SxSI driver; unlike piscsi trace logging is not a runtime option, but you have to edit a C++ source file and recompile:

in RaSCSI/source/raspberrypi/disk.cpp fine these lines and remove the comment slashes

```c++
//#define DISK_LOG
//#define DISK_LOG_WARNING
```

if you have a build environment on your RPi, `make rascsi` should trigger the compiler

then with the recompiled binary, redirect stdout to a file for convenience, e.g.

```
./rascsi -HD0 /var/rascsi/SASI-40MB-MasterDiskV3.HDF 2>&1 | tee rascsi-x68000.log
```

would you be okay to attempt either or both?

[12 more replies]

closed 2026-08-30 by rdmark