PiSCSI/piscsi #814: Explicit SCSI-1 support: Add support for .hd1 image files

uweseimet, 2022-08-26:
rascsi shall support a new image type ".hd1". This type is identical with ".hds" except for the SCSI level and response code format being SCSI-1 instead of SCSI-2.
The alternative approach of being able to (having to ...) configure the SCSI and response code levels would force additional configuration settings on the users and would require much more changes in the core, and also in the clients. Most users do not have the expert knowledge required for these settings anyway and would run into issues with regular platforms when playing with these settings.

Also see https://github.com/akuker/RASCSI/issues/799.

uweseimet, 2022-09-03:
@Pacjunk In order to speed things up a bit I created a branch testing_scsi1. This branch contains all pending changes (SASI removal) plus the changes required to support the .hd1 extension. Can you please test whether the testing_scsi1 branch resolves the remaining issues with your VAXstation?

uweseimet, 2022-09-03:
@Pacjunk There is a branch feature_scsi1_image_file now, which will be the branch used for the upcoming PR.

uweseimet, 2022-09-03:
@akuker You have changed the c-cpp.yml file in this branch. I'm not sure whether this was intentional, because there is also a ci_integration branch.

akuker, 2022-09-03:
> @akuker You have changed the c-cpp.yml file in this branch. I'm not sure whether this was intentional, because there is also a ci_integration branch.

I was trying to keep it from spamming your email inbox as much :) Since I enabled the action, it's going to fail every time you commit.

uweseimet, 2022-09-03:
@akuker I see, thank you. I was not aware was this file was doing, this why I was asking. I assume we have to ensure that it is not merged into develop later?

akuker, 2022-09-03:
You are correct. Feel free to take a look at what I've got at : https://github.com/akuker/rascsi_ci_integration/actions 

I'll merge it into the real repo later today

uweseimet, 2022-09-03:
@akuker I will most likely have a look at the actions in the future. I'm not familiar with GitHub actions because I use GitLab for my own projects that use automation. GitLab was the better choice for me because I needed a free Docker repository for them. GitHub did not provide a free *private* repository when I set up my projects. As far as I know they still do not provide that.

Pacjunk, 2022-09-04:
@uweseimet OK, I have tested on the new branch!  The web interface is broken, so I had to learn how to use rasctl (not a bad thing!). Anyway, tested with an hds file and an hd1 file. The hds file returned levels 2/2 and the hd1 file returned levels 1/1 so that bit worked OK.

However, I noticed that my system would now boot off the hds file (SCSI-2) as well. After looking at the trace logs (more rasctl commands!), it appears that the code is no longer preventing page code 0, so my system boots fine off the SCSI-2 device. Was this something missed during your major code rearrangement? - not that I'm complaining :-)

uweseimet, 2022-09-04:
@Pacjunk :-) I'm afraid this is a bug, mode page 0 should still be rejected. There is no unit test yet for this. I cannot immediately see from the code what's wrong, maybe you can? Adding a log with the relevant MODE SENSE would help.

Regarding non-existent mode pages in general it would be possible to extend rascsi, so that you can specify static content for a mode page on the command line, by providing the sequence of bytes to be returned. But I would not really like to add this functionality. The clients would also have to offer this, making it even harder to understand all available settings. In addition, you would have to know which data the device has to return for a particular page. For vendor-specific pages you usually do not know that.

uweseimet, 2022-09-04:
@Pacjunk After having a closer look at the code I would expect the condition in mode_page_device.cpp:50:

```
	LOGTRACE("%s Requesting mode page $%02X", __PRETTY_FUNCTION__, page);

	// Mode page data mapped to the respective page numbers, C++ maps are ordered by key
	map<int, vector<BYTE>> pages;
	AddModePages(pages, page, changeable);

	if (pages.empty()) {
		LOGTRACE("%s Unsupported mode page $%02X", __PRETTY_FUNCTION__, page);
		throw scsi_error_exception(sense_key::ILLEGAL_REQUEST, asc::INVALID_FIELD_IN_CDB);
	}
```
to be true (page list is empty) and add a trace log item and throw an exception, which would result in an error condition to be reported.

uweseimet, 2022-09-04:
@Pacjunk I just noticed I was wrong and there already is a unit test for page 0. It's the first in mode_pages_test.cpp. It tests that an exception is raised if page 0 is requested.

Pacjunk, 2022-09-04:
Attached is part of the log. That "Requesting mode page" message isn't even there!

```
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.743] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Selection() Selection Phase Target ID=2
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() Command Phase
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() CDB[0]=$1A
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() CDB[1]=$00
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() CDB[2]=$00
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() CDB[3]=$00
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() CDB[4]=$0C
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] virtual void ScsiController::Command() CDB[5]=$00
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [trace] void ScsiController::Execute() Execution phase command $1A
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.744] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = ModePageDevice; DWORD = unsigned int] Executing ModeSense6 ($1A)
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] virtual void ScsiController::DataIn() Going into Data-in Phase
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] void ScsiController::Send() Sending handhake with offset 0, length 12
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] void ScsiController::Send() Move to next phase datain (6)
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] virtual void ScsiController::Status() Status Phase $00
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.745] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] void ScsiController::Send() Move to next phase status (8)
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] virtual void ScsiController::MsgIn() Message In phase
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 6
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] void ScsiController::Send() Move to next phase msgin (9)
Sep  4 17:40:57 raspberrypi RASCSI[433]: [2022-09-04 17:40:57.746] [trace] virtual void ScsiController::BusFree() Bus free phase

```

uweseimet, 2022-09-04:
Can you please investigate?

uweseimet, 2022-09-04:
@Pacjunk Hmm, the previous comment is strange. It resulted from a mail reply, but some github did not deal well with it ;-).

Anyway, what I mean is that there appears to be a contradition: Either mode page 0 is returned even though it shouldn't be, and in this case the trace log should show the respective MODE SENSE command and how it is processed. Or the trace log shows nothing, and in this case there wouldn't have been any MODE SENSE command at all, also not for page 0.
Note sure what the resolution is of this contradiction is, but this is what I would like you to investigate. Please always use the latest sources from the feature branch.

Pacjunk, 2022-09-06:
In disk.cpp:

`int length = (int)cdb[4];`  --> The modesense6 requests length 12 (in my trace)
`size = 12;`  --> Length of the mode parameter block
`size += super::AddModePages(cdb, &buf[size], length - size);` --> length-size = 12 - 12 = 0

In mode_page_device.cpp:33:

```
int ModePageDevice::AddModePages(const DWORD *cdb, BYTE *buf, int max_length)
{
	if (max_length <= 0) {
		return 0;
	}
```
In the request my machine sends, max_length **will** be 0. This is <=0 so the test will exit the function immediately. That is why we don't see the "requesting mode page" message.

Zero is a valid value, so maybe the test should be <0 instead of <=0. Then again, is it **really** possible for max_length to be negative??  If not, drop the whole "if" statement.

uweseimet, 2022-09-06:
@Pacjunk I added an explicit test for mode page 0 to my external test sute, which is running on my Atari. The mode page part (see below) also does not find anything wrong. It can read all available mode pages, and trying to read the non-existing mode page 0 fails because there is none. From my perspective this is consistent with the unit tests and the implementation and is the correct behavior.
If there is still anything wrong on your side, feedback would be great.

```
  MODE SENSE
    Reading all mode pages with MODE SENSE (6)
      Received 106 data bytes
        Available pages list: 1, 3, 4, 8, 48
        Page 3: Format device page (current, not savable)
          Tracks per zone: 8
          Alternate sectors per zone: 0
          Alternate tracks per zone: 0
          Alternate tracks per logical unit: 0
          Sectors per track: 25
          Data bytes per physical sector: 512
          Interleave: 1
          Track skew factor: 11
          Cylinder skew factor: 20
          Soft sector formatting (SSEC): 0
          Hard sector formatting (HSEC): 1
          Removable (RMB): 0
          Surface (SURF): 0
        Page 4: Rigid disk drive geometry page (current, not savable)
          Number of cylinders: 23714
          Number of heads: 8
          Starting cylinder-write precompensation: 0
          Starting cylinder-reduced write current: 0
          Drive step rate: 0
          Landing zone cylinder 0
          Rotational offset: 0
          Medium rotation rate: 7200
        Page 8: Caching page (current, not savable)
          Read cache disable (RCD): 0
          Write cache enable (WCE): 0
          Multiplication factor (MF): 0
          Write retention priority: 0
          Demand read retention priority: 0
          Disable pre-fetch transfer length: -1
          Minimum pre-fetch: 0
          Maxium pre-fetch: -1
          Maximum pre-fetch ceiling: -1
    Reading all mode pages with MODE SENSE (10)
      Received 110 data bytes
        Available pages list: 1, 3, 4, 8, 48
    Checking for support of vendor-specific mode page 0
      Mode page 0 is not supported
```

uweseimet, 2022-09-06:
@Pacjunk My comment overlaps with yours. I will check.

uweseimet, 2022-09-06:
@Pacjunk I see. I don't want to remove the statement because AddModePages is called from different locations and max_value is a signed value, and I want to keep is a part of defensive programming.
But I just commited your suggested change regarding the max_length check to the feature branch. Can you please verify?

uweseimet, 2022-09-06:
@Pacjunk By the way, a user of my hard disk software sent me information that shows that BlueSCSI drives are SCSI-1, not SCSI-2:
```
 INQUIRY
    Calling with valid data
      Device type: Direct Access
      Removable media support: No
      Device name: 'QUANTUM'
      Firmware revision: '1.1'
      SCSI/SPC version: SCSI-1 CCS
      Response data format: SCSI-1 CCS
      Additional length: $1F
```
This can explain some differences between RaSCSI and BlueSCSI. Even with .hd1 files there might be cased where rascsi is not SCSI-1 compliant. If such a case is found a new ticket should be created. The current change only deals with INQUIRY values and nothing else.

Pacjunk, 2022-09-06:
That should do it, but I can't really test for a day or so  (sorry about the earlier delay - I was away). I'll let you know.

Yes, I know bluescsi returns SCSI-1 for the inquiry - that's why I never had the problem with that system. My system worked fine with the .hd1 image file, so it only seems to care about the inquiry results.

uweseimet, 2022-09-06:
@Pacjunk I don't expect that regarding SCSI-1 anybody is going to test more right now. What I meant is that if things are fine for you now I can raise a PR, because this ticket is for .hd1 only anyway. If somebody should find anything wrong with SCSI-1 in the future, she/he can create a new ticket. WRT to SCSI-1 (other than INQUIRY) the current code does not differ from previous releases anyway. So any problem would very likely have been there before.
Please confirm that we are fine for the final PR.

Pacjunk, 2022-09-06:
OK. I'm rebuilding now, but won't be able to test until tomorrow night. I'll let you know in 24 hours.  Thanks,

uweseimet, 2022-09-06:
@Pacjunk OK, thank you. If a different issue than the one this ticket is directly dealing with should pop up, a new ticket should be created, so that we can more this .hd1 change.
After testing, please also let me know whether https://github.com/akuker/RASCSI/issues/799 is resolved with this ticket, or whether there is still something left in https://github.com/akuker/RASCSI/issues/799 that needs to be addressed. Hopefully not ;-).

Pacjunk, 2022-09-07:
@uweseimet OK, things seem to be operating as expected.

If I try to boot a .hds file, then the system errors out and RASCSI generates the following:

```
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.936] [trace] virtual void ScsiController::Command() Command Phase
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.936] [trace] virtual void ScsiController::Command() CDB[0]=$1A
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.936] [trace] virtual void ScsiController::Command() CDB[1]=$00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.936] [trace] virtual void ScsiController::Command() CDB[2]=$00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] virtual void ScsiController::Command() CDB[3]=$00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] virtual void ScsiController::Command() CDB[4]=$0C
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] virtual void ScsiController::Command() CDB[5]=$00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] void ScsiController::Execute() Execution phase command $1A
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = ModePageDevice; DWORD = unsigned int] Executing ModeSense6 ($1A)
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] int ModePageDevice::AddModePages(const DWORD*, BYTE*, int) Requesting mode page $00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] int ModePageDevice::AddModePages(const DWORD*, BYTE*, int) Unsupported mode page $00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [debug] Error status: Sense Key $05, ASC $24, ASCQ $00
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] virtual void ScsiController::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Sep  7 17:39:50 raspberrypi RASCSI[434]: [2022-09-07 17:39:50.937] [trace] virtual void ScsiController::Status() Status Phase $02

```

This is the expected result.  If I boot the .hd1 file, then everything works fine. Once the system is booted I can access both drive types as the page 0 problem only occurs at boot. I'm happy for the solution to progress  (not so much that a solution is required in the first place, but hey, that's life!). Thanks for your work on this.

@rdmark Does the web front end need any tweaking to support the .hd1 image files?  I can't look myself as the web server seems to be broken in this branch.

uweseimet, 2022-09-07:
@Pacjunk Thank you for your feedback and the time spent on testing! I am going to finalize the PR.

uweseimet, 2022-09-07:
@Pacjunk Please remember to comment on the status of https://github.com/akuker/RASCSI/issues/799. Is there anything left in https://github.com/akuker/RASCSI/issues/799 to be done after merging the .hd1 support?

Pacjunk, 2022-09-07:
No, should be able to close that one.

rdmark, 2022-09-07:
If the web UI won’t start I guess there must be some assumption in the code
that isn’t valid anymore. I can take a look when I’m back from vacation in
two weeks, unless someone else beats me to it. ;)

Also at the very least the hd1 image type should be added to the image file
creation drop down.

closed 2022-09-08 by uweseimet