PiSCSI/piscsi #818: MODE SELECT for format page (page 3) is not working

uweseimet, 2022-08-29:
When resolving https://github.com/akuker/RASCSI/issues/815 it turned out that something is wrong with the handling of the parameters passed to MODE SELECT. rascsi reports an error when trying to change the sector size, whereas a real SCSI drive that supports a sector size change (FUJITSU M2624S) does not.
The error appears to be related to the evaluation of the block descriptor data length.

uweseimet, 2022-10-08:
After fixing the descriptor and offset handling rascsi behaves like the Fujitsu drive and a MODE SELECT/FORMAT sequence works as far as rascsi can support it.
```
[2022-10-08 13:07:25.373] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
[2022-10-08 13:07:25.373] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = ModePageDevice] Executing ModeSense6 ($1A)
[2022-10-08 13:07:30.382] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
[2022-10-08 13:07:30.382] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = ModePageDevice] Executing ModeSense6 ($1A)
[2022-10-08 13:07:30.384] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $15
[2022-10-08 13:07:30.384] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = ModePageDevice] Executing ModeSelect6 ($15)
[2022-10-08 13:07:30.421] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $04
[2022-10-08 13:07:30.421] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = Disk] Executing FormatUnit ($04)

2022-10-08 13:08:47.508] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = ModePageDevice] Executing ModeSense6 ($1A)
[2022-10-08 13:08:54.682] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
[2022-10-08 13:08:54.682] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = ModePageDevice] Executing ModeSense6 ($1A)
[2022-10-08 13:08:54.684] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $15
[2022-10-08 13:08:54.684] [debug] bool Dispatcher<T>::Dispatch(T*, scsi_defs::scsi_command) [with T = ModePageDevice] Executing ModeSelect6 ($15)
[2022-10-08 13:08:54.684] [warning] In order to change the sector size use the -b option when launching rascsi
[2022-10-08 13:08:54.685] [debug] Error status: Sense Key $05, ASC $26

```

rdmark, 2022-10-09:
@uweseimet FYI I've adjusted the long-lived branch filter on Sonar to capture 'develop' as well as anything matching 'feature-.*'

In order for develop to be treated a s long-lived branch, I had to delete the current instance of the develop analysis on Sonar, which is why you may notice that it's gone from the branch list. Once Sonar picks up the branch again, it should be treated as long-lived.

closed 2022-10-10 by uweseimet