PiSCSI/piscsi #821: Compilation fails on Debian Bullseye amd64 (gcc version 10.2.1)

nucleogenic, 2022-09-01:
# Info

- Which version of Pi are you using: N/A
- Which github revision of software: e649a01
- Which board version: N/A
- Which computer is the RaSCSI connected to: None

# Describe the issue

Compilation fails on Debian Bullseye amd64 (gcc version 10.2.1) with the error messages below.

(This relates to adding Bullseye support to https://github.com/akuker/RASCSI/pull/819.)

```
#12 22.03 g++ -O3 -Wall -Werror -DNDEBUG -Wno-psabi -std=c++17 -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP  -DFMT_HEADER_ONLY -DCONNECT_TYPE_FULLSPEC -c ./devices/scsi_printer.cpp -o obj/fullspec/scsi_printer.o
#12 26.31 ./devices/cfilesystem.cpp: In member function 'void CHostFilename::SetEntryName()':
#12 26.31 ./devices/cfilesystem.cpp:1094:20: error: writing 1 byte into a region of size 0 [-Werror=stringop-overflow=]
#12 26.31  1094 |    m_dirHuman.add[i] = *p++;
#12 26.31       |    ~~~~~~~~~~~~~~~~^
#12 26.31 In file included from ./devices/cfilesystem.cpp:23:
#12 26.31 ./devices/cfilesystem.h:207:8: note: at offset [2, 10] to object 'Human68k::dirent_t::add' with size 10 declared here
#12 26.31   207 |   BYTE add[10];   ///< +12 File name addition (PADDING 0x00)
#12 26.31       |        ^~~
```

**Replication steps:**

1. Install Debian Bullseye amd64 on a VM
2. apt install git sudo
3. Add pi user (in sudo group)
5. Switch to pi user
6. git clone https://github.com/akuker/RASCSI.git --branch develop
7. cd RASCSI && ./easyinstall.sh
8. Select option 10 (standalone RaSCSI)

Alternatively:

```
git clone https://github.com/nucleogenic/RASCSI.git --branch docker-dev-environment
cd RASCSI/docker
DEBIAN_ARCH=amd64 DEBIAN_VERSION=bullseye docker compose build --progress=plain rascsi
```

I also tried Debian Bullseye arm32v7 (gcc 10.2.1) and Ubuntu 22.04 amd64 (gcc 11.2.0), but I was unable to replicate the issue.

Any help, hints or advice would be much appreciated.

uweseimet, 2022-09-01:
@nucleogenic Your platform is not a supported build platform. I'm afraid I therefore have to reject this ticket, because we cannot test building on any available Linux platform. Supported platforms are Raspian and Ubuntu.
IMHO the error message is wrong. In case you think ithe message is correct and you can fix the problem, please do so and create a PR.

closed 2022-09-01 by uweseimet

akuker, 2022-09-03:
We could probably look at cleaning up that file at some point. The Sharp X68000 specific host device hasn't been tested in a long, long, long time, and at this point, we don't have any way to test or support it. So... any functionality that just supports the Human OS could be removed (IMO)

uweseimet, 2022-10-02:
Even though I closed this ticket I have not forgotten that it exists ;-). The problem the compiler reports (only on this platform it seems, but whether gcc warns about something or not appears to change with gcc versions) reminds me of some of the SonarCloud issues reported for cfilesystem.cpp.
I am probably not good enough in juggling with pointers, because for some of the reported issues I could not figure out whether the code is right or wrong, i.e. whether a pointer might point to unallocated memory or not. But usually SonarCloud is right and there are code issues, indeed.
The complete Human OS code is a mess, and it is responsible for almost all remaining SonarCloud issues.

uweseimet, 2022-10-16:
@nucleogenic And yet one more comment: Have you already tried clang++ instead of g++? On my Linux PC I switched from g++ to clang++ for compiling RaSCSI because clang++ is about 30% faster, which is a lot. clang++ might accept the offending code.

nucleogenic, 2022-11-11:
@uweseimet Now compiling OK on Bullseye with clang++, so the Docker patch file workaround is no longer needed. This will be removed shortly 😎