PiSCSI/piscsi #825: Can't upload disk images with ending ".img" via web interface

fdanapfel, 2022-09-02:
# Info

- Which version of Pi are you using: Zero 2 W
- Which github revision of software: v22.08.01
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Atari TT030

# Describe the issue
When trying to upload a disk image that has the ending ".img" via the web-interface it always gives the message "You can't upload files of this type."., but according to https://github.com/akuker/RASCSI/pull/205 .img should be a valid suffix.

Changing the ending from .img to .hds works, but it is a bit annoying to have to do that for each disk image I want to use with RASCSI.

I'm using Safari 15.6.1 on macOS 10.15.7 as the web-browser.

rdmark, 2022-09-02:
So there is some nuance to this issue. TL;DR; 

First off, the Web Interface simply inquires the RaSCSI backend which file endings that it recognizes, and img is not one that is recognized currently. One reason for this is that img can be any kind of image really. There's no telling a hard drive img from a floppy img from a CD-ROM img, etc. RaSCSI requires there to be a clear connection between the file ending and the type of device that is imaged.

Here's how to ask the rascsi backend which file types it supports:
```
$ rasctl -m
Supported image file extension to device type mappings:
  hda->SCHD
  hdf->SAHD
  hdi->SCHD
  hdn->SCHD
  hdr->SCRM
  hds->SCHD
  iso->SCCD
  mos->SCMO
  nhd->SCHD
```

Secondly, we have applied the file upload file ending validation because rascsi continuously scanning the images/ dir, and will log warnings when it encounter files that it doesn't consider valid image files (the authentication is not based on file endings, but on file size being a multiple of 512 B.) Having a ton of invalid files in the images/ dir may quickly make things messy for the user.

All that said, I can see an argument for dropping the upload file ending validation in the Web Interface. Instead of erecting obstacles for the user, entrust them to be careful with which files they upload. Let me request comments from the community and hear if someone has an argument for another approach.

nucleogenic, 2022-09-02:
I have found the restriction unhelpful in a couple of cases, such as uploading a `.toast` file.

The treatment of files, once uploaded, does not seem to be restricted, e.g. I can attach an file with a `.jpg` extension.

I would be happy to have the extension check removed, or alternatively the list of accepted files expanded.

uweseimet, 2022-09-03:
@fdanapfel I just merged @rdmark 's changes. Can you please check and close this ticket if your issue is resolved?

fdanapfel, 2022-09-03:
@rdmark Thanks a lot for the quick response and explanation of the reasoning why this happens. And also for agreeing to change this based on the comments from @nucleogenic.

@uweseimet If you can provide some guidance on how to switch the installed RaSCSI version to the development branch I will gladly test the changes and report back my findings.

uweseimet, 2022-09-03:
@fdanapfel In order to test you should clone the git repository and then run "git checkout develop". Now you can run easyinstall as usual to install/update RaSCSI.
If you are not prepared to use git, you can download a ZIP archive with the develop branch. Go to https://github.com/akuker/RASCSI/tree/develop, and after clicking on the "Code" button you will find a "Download ZIP" link.

fdanapfel, 2022-09-03:
@uweseimet Thanks for the guidance. After switching to the development branch using "git checkout develop" and then using easy install to update EaSCSI and the web interface it is now possible to upload files ending with .img.

@rdmark Big thanks to  you for changing this so quickly!

Issue is resolved, so as requested I'm closing this ticket.

closed 2022-09-03 by fdanapfel