PiSCSI/piscsi #830: Removal of X68000 support

nucleogenic, 2022-09-03:
This issue is to propose the removal of X68000 support from RaSCSI Reloaded.

It is unknown whether this functionality actually works and there is currently no known means to test it.

It would be useful hear from any users who have experience with X68000 and RaSCSI Reloaded - please leave a comment.

crivett95, 2022-09-04:
I have an X68000 Expert (a SASI model) that I have the sxsi driver loaded on to enable SCSI support. I received the machine with a SCSI2SD fitted internally but have recently removed this in favour of using RaSCSI Reloaded (which I have been enjoying on other machines) from the external port on the back of the machine. It works fine as a normal SCSI HDD so far. I haven't tried any of the X68000 specific features yet but had been planning to. I would be happy to test these and help out in any way I can in terms of X68000 support.

rdmark, 2022-09-04:
@crivett95 Thank you for volunteering with testing on your X68k! There are two X68k specific features, both enabled through the "Bridge" device: a Neptune-X compatible ethernet adapter (similar to the DaynaPORT ethernet device for Macs and Ataris), and the Host File System driver, which allows you to mount the host's Linux file system in the Human OS for easy file sharing between the two.

I've translated some of the Japanese documentation over at this RaSCSI fork repo: https://github.com/rdmark/RASCSI-X68k/wiki

We don't distribute the pre-compiled Human OS software in the Reloaded repo anymore, but you can find them over at the RaSCSI-X68k repo: https://github.com/rdmark/RASCSI-X68k/tree/main/bin/x68k

Please have a look and let us know how it goes!

crivett95, 2022-09-04:
@rdmark No problem! I've had a read through some of the docs and will try find some time to test both features over the next week or so. If I can get the pre-compiled binaries working, I'll see about setting up a C compiler to try out your translation effort.

crivett95, 2022-09-10:
I've managed to get the files onto my X68K to give this a try. I'm unsure about any extra networking bits I need to load onto the X68K for the RASETHER.SYS driver to work, but have tried the RASDRV.SYS driver and the RASCTL.X utility. Unfortunately, RASDRV doesn't load and RASCTL reports that the bridge is not found. Having had a quick look at the source for these it looks like they are looking for a product id of "RASCSI BRIDGE". The X68000 host bridge seems to have a product id of "SCSI HOST BRIDGE" instead according to the web ui. I suspect this may be causing the issue? I've attached some screen captures from the X68K running through my OSSC into an HDMI capture device.

[image]

[image]

rdmark, 2022-09-22:
@crivett95 I put up a tentative PR to change the product name back to RASCSI BRIDGE, see https://github.com/akuker/RASCSI/pull/851. Would you be able to build and test if this addresses the issue?

crivett95, 2022-09-22:
@rdmark Sure, I should have some time to give it a try and report back either tomorrow or over the weekend.

crivett95, 2022-09-24:
@rdmark I've just built and tested your branch with the product name changed back to RASCSI BRIDGE. It has changed something as now when the X68K starts up it always gives a read error that goes away on retry anytime the RASCSI BRIDGE is attached via the web interface. Other than the appearance of this read error, it behaves the same as before though unfortunately. I'm wondering if it might be worth me trying the original pre-fork RaSCSI to confirm I can get it working there on my X68K since compatibility when using the SxSI driver for SCSI support was never fully tested on SASI models according to the translated documentation.

[image]

rdmark, 2022-09-24:
Yes it is definitely worth trying with the last GIMONS version of the RaSCSI software. You can find it here https://github.com/rdmark/RASCSI-X68k

ajacocks, 2022-10-18:
I'll see if I can test this on my X68000 CZ-600C (the original model). Hopefully this weekend.

rdmark, 2022-10-18:
@ajacocks Do you have a SCSI controller for your original X68k, or will you be using SASI?

rdmark, 2022-11-05:
User Will.Broke.It on Discord has confirmed that the Host Bridge works with the RASETHER driver on their X68k to the stage that they were able to ping other servers.

[image]

ajacocks, 2022-11-06:
@rdmark I do not, and plan to use SASI.

rdmark, 2022-11-12:
@ajacocks Got it. For your testing, please use the August 2022 release of the RaSCSI software. SASI support was removed in October 2022. See https://github.com/akuker/RASCSI/issues/809 for more info.

akuker, 2023-09-23:
> This issue is to propose the removal of X68000 support from RaSCSI Reloaded.

Closing this issue. I now have a X68000 machine and will be working on testing these features in the coming months.

closed 2023-09-23 by akuker

koshisan, 2025-06-20:
Hi! I was trying to read up on this - but it is a bit unclear what the current status of RASDRV.SYS and RASETHER.SYS is...
I can't seem to get neither working on my X68000 XVI. The RASDRV says that no drive is found (I suspect "Host Services" are the correct "device" to mount? The documentation talks about a "bridge", but those seem to be network related...). 
RASETHER.SYS seems to hang the machine...