PiSCSI/piscsi #831: [QUESTION]

Rowan-Bird, 2022-09-03:
# Info

Hello, I am here to ask the following:
Can I use an original Raspberry Pi (this assumes my Pi is functional) with the small GPIO interface
Can this device emulate an A/UX compatible CD-ROM drive

rdmark, 2022-09-04:
@Rowan-Bird The original RPi with a 26 pin GPIO header is not supported. See https://github.com/akuker/RASCSI/wiki/Compatibility#Raspberry_Pi_compatibility

If you can tell us what is required for a CD-ROM drive to be A/UX compatible, we may be able to comment on this. Early Apple CD-ROM drivers for Mac expect non-standard page mode responses which RaSCSI doesn't support yet. See issue #1.

For further questions, I suggest you create a thread in Discussions, or head over to our Discord server.

closed 2022-09-04 by rdmark

Rowan-Bird, 2022-09-04:
I haven't read that, sorry for wasting your time.
As for A/UX CD-ROMs, I was asking if it could emulate a drive compatible with A/UX (e.g. Apple CD-150/300)

rdmark, 2022-09-04:
@Rowan-Bird If you look at issue #1 you can see that there is some nuance to RaSCSI's support for Apple CD-150, 300 etc. At least for Mac OS, Apple's drivers are expecting non-standard page mode responses from the hardware. If the A/UX drivers are the same, you will have problems. On Mac, we've worked around the problem by hacking Apple's drivers to ignore the page mode check.