PiSCSI/piscsi #836: Run disktype from web UI

nucleogenic, 2022-09-07:
# Overview

The `disktype` utility was [recently added](https://github.com/akuker/RASCSI/pull/820) to the RaSCSI installer.

```
pi@rascsi:~/images $ disktype RaSCSI-Boot-7.0.1.hda

--- RaSCSI-Boot-7.0.1.hda
Regular file, size 40 MiB (41943040 bytes)
Apple partition map, 3 entries
Partition 1: 31.50 KiB (32256 bytes, 63 sectors from 1)
  Type "Apple_partition_map"
Partition 2: 16 KiB (16384 bytes, 32 sectors from 64)
  Type "Apple_Driver"
Partition 3: 39.95 MiB (41893888 bytes, 81824 sectors from 96)
  Type "Apple_HFS"
  HFS file system
    Volume name "RaSCSI 7.0.1"
    Volume size 39.94 MiB (41884672 bytes, 40903 blocks of 1 KiB)
```

It would be convenient to run this command and show its output from the web UI.

# Proposal
 
**(i)** Add an 'Info' button to each row in the image management table, except where the file has an archive extension.

[image]

**(ii)** Add a screen that shows the `disktype` command output. This should not use the 'flash' mechanism (which is better suited to simple status messages) due to the volume of output and formatting requirements.

[image]

akuker, 2022-09-07:
Thank you for writing this up!

rdmark, 2022-09-08:
Random thought: a solution here for displaying shell output nicely
formatted in a jinja template could be potentially reused for the syslog
output as well.

rdmark, 2022-09-25:
PR ready for review https://github.com/akuker/RASCSI/pull/863

@nucleogenic Heads-up that this conflicts mildly with your changes in https://github.com/akuker/RASCSI/pull/846 . I'll update my PR once your changes have been merged.

closed 2022-09-27 by rdmark