PiSCSI/piscsi #841: Internal Server Error when reattaching ISO image

Pacjunk, 2022-09-09:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: develop
- Which board version:FULLSPEC
- Which computer is the RaSCSI connected to: n/a

# Describe the issue
If I attach an ISO image, then click the "Eject" button I get to the following (as expected) :
[image]
If I then click the "Attach" button, RASCSI generates an error and all disks are detached:
[image]
I then need to reload my default configuration to get back to normal. The same operation seems to work fine with a Removable disk, only CD/DVD seems to be affected.

rdmark, 2022-09-09:
Very interesting, I’m pretty sure this used to work until recently.
Are you able to reproduce this using rasctl, taking the Web UI itself out
of the equation?

Pacjunk, 2022-09-09:
Didn't know rasctl could do that. I will have a look...

Pacjunk, 2022-09-09:
Yes, looks like rasctl does it too. You are excused!

```
pi@raspberrypi:~ $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  2 |   0 | SCHD | /home/pi/images/boot72.hds
|  3 |   0 | SCHD | /home/pi/images/DEC_RZ28M.hds
+----+-----+------+-------------------------------------
pi@raspberrypi:~ $ rasctl -i 1 -c attach -f alpha084.iso
pi@raspberrypi:~ $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  1 |   0 | SCCD | /home/pi/images/alpha084.iso (READ-ONLY)
|  2 |   0 | SCHD | /home/pi/images/boot72.hds
|  3 |   0 | SCHD | /home/pi/images/DEC_RZ28M.hds
+----+-----+------+-------------------------------------
pi@raspberrypi:~ $ rasctl -i 1 -c eject
pi@raspberrypi:~ $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  1 |   0 | SCCD | NO MEDIA
|  2 |   0 | SCHD | /home/pi/images/boot72.hds
|  3 |   0 | SCHD | /home/pi/images/DEC_RZ28M.hds
+----+-----+------+-------------------------------------
pi@raspberrypi:~ $ rasctl -i 1 -c insert -f alpha084.iso
Error:
pi@raspberrypi:~ $ rasctl -l
No images currently attached.

```

uweseimet, 2022-09-09:
I will have t a look at this, but I would like to wait until after https://github.com/akuker/RASCSI/pull/840 has been merged.

uweseimet, 2022-09-09:
@Pacjunk The fix_issue_841 branch should fix the problem. In addition, it improves memory management and removes the static limit of 96 tracks.

Pacjunk, 2022-09-09:
Yes, that fixed it. Thanks

closed 2022-09-15 by uweseimet