PiSCSI/piscsi #844: [ASR-10] Can't mount

lukexi, 2022-09-10:
# Info

- Which version of Pi are you using: 3B+
- Which github revision of software: f0c36f
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: Ensoniq ASR-10

# Describe the issue
Attempting to mount a SCSI CD-ROM (an Ensoniq Sample CD ISO, via Command>Directory>Change Storage Device on the ASR) fails with this error
```
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.566] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.566] [trace] virtual void ScsiController::Selection() Selection Phase Target ID=3
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.566] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() Command Phase
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() CDB[0]=$03
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() CDB[1]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() CDB[2]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() CDB[3]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() CDB[4]=$0E
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Command() CDB[5]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] void ScsiController::Execute() Execution phase command $03
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $03
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing RequestSense ($03)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] std::vector<unsigned char> PrimaryDevice::HandleRequestSense() Status $00, Sense Key $00, ASC $00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::DataIn() Going into Data-in Phase
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] void ScsiController::Send() Sending handhake with offset 0, length 14
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] void ScsiController::Send() Move to next phase datain (6)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::Status() Status Phase $00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] void ScsiController::Send() Move to next phase status (8)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::MsgIn() Message In phase
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.567] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] void ScsiController::Send() Move to next phase msgin (9)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::BusFree() Bus free phase
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Selection() Selection Phase Target ID=3
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() Command Phase
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() CDB[0]=$12
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() CDB[1]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() CDB[2]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() CDB[3]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() CDB[4]=$24
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::Command() CDB[5]=$00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] void ScsiController::Execute() Execution phase command $12
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $12
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing Inquiry ($12)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual void ScsiController::DataIn() Going into Data-in Phase
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.568] [trace] void ScsiController::Send() Sending handhake with offset 0, length 36
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] void ScsiController::Send() Move to next phase datain (6)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] virtual void ScsiController::Status() Status Phase $00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is unknown
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.569] [trace] void ScsiController::Send() Move to next phase status (8)
[... cut ...]
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.573] [debug] Error status: Sense Key $05, ASC $20, ASCQ $00
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.573] [trace] virtual void ScsiController::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.573] [trace] virtual void ScsiController::Status() Status Phase $02
Sep 10 22:26:39 seascape-seastone RASCSI[5527]: [2022-09-10 22:26:39.573] [trace] virtual void ScsiController::BusFree() Bus free phase
```

lukexi, 2022-09-10:
Attempting to rename the .iso to .hds as suggested in the Wiki (https://github.com/akuker/RaSCSI/wiki/RaSCSI-for-Hardware-Samplers#compatibility) works! But is it worth keeping this open to debug why it won't work as a CD-ROM image?

lukexi, 2022-09-10:
Ah, I spoke too soon. It "works" in that I can at least browse the top level directories, but attempting to enter one on the ASR results in:
```
[2022-09-10 22:33:22.674] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-10 22:33:22.674] [trace] virtual void ScsiController::Selection() Selection Phase Target ID=1
[2022-09-10 22:33:22.674] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-10 22:33:22.674] [trace] virtual void ScsiController::Command() Command Phase
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Command() CDB[0]=$00
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Command() CDB[1]=$00
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Command() CDB[2]=$00
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Command() CDB[3]=$00
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Command() CDB[4]=$00
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Command() CDB[5]=$00
[2022-09-10 22:33:22.675] [trace] void ScsiController::Execute() Execution phase command $00
[2022-09-10 22:33:22.675] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $00
[2022-09-10 22:33:22.675] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing TestUnitReady ($00)
[2022-09-10 22:33:22.675] [trace] void PrimaryDevice::CheckReady() Device is ready
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::Status() Status Phase $00
[2022-09-10 22:33:22.675] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-10 22:33:22.675] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
[2022-09-10 22:33:22.675] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-10 22:33:22.675] [trace] void ScsiController::Send() Move to next phase status (8)
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::MsgIn() Message In phase
[2022-09-10 22:33:22.675] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
[2022-09-10 22:33:22.675] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
[2022-09-10 22:33:22.675] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
[2022-09-10 22:33:22.675] [trace] void ScsiController::Send() Move to next phase msgin (9)
[2022-09-10 22:33:22.675] [trace] virtual void ScsiController::BusFree() Bus free phase
```

lukexi, 2022-09-10:
(And a "File Operation Error" message on the ASR)

lukexi, 2022-09-10:
@chickeneps I've seen you say the ASR is working in the RaSCSI Discord, do you have a description of what worked for you? I haven't had any luck with either CD-ROMs or formatting disks...

chickeneps, 2022-09-10:
I attach like this: rasctl -i 4 -c attach -t hd -f /home/pi/images/myimage.hda

myimage.hda is 2GB large, nothing more, but it should work on anything less.

I plug the RaSCSI via M/M DB25 adapter and termination is on, on the RaSCSI. I've used Pi Zero 1.1, 1.3, 2, and Pi 3. All test thoroughly. My test is 1) do a format from the ASR-10, save the OS, and write the default directories. Then I power cycle everything and see if the ASR-10 boots off the RaSCSI. Then, I COMMAND STORAGE DEVICE to a valid Ensoniq image and load large files (like 4000 blocks) and save them to the newly formatted image.

Otherwise, nothing special.

rdmark, 2022-09-22:
@lukexi Did you get this to work with the last piece of advice?

uweseimet, 2022-11-03:
@lukexi Can you please provide feedback, based on the current develop branch? Your first log shows that the MODE SELECT command reported an error. (You second log does not show any error condition that was logged.) The MODE SELECT implementation was updated recently.
If there is still an issue, please provide an updated rascsi log, but only if the log clearly reports an error. A log without an error message on the rascsi side is unlikely to help with resolving your problem.

lukexi, 2022-11-10:
I rebuilt from the latest develop branch and retried loading a CD and it failed the same FILE OPERATION ERROR — I'll capture another log.

@chickeneps @rdmark hmmm yeah, I just tried the @chickeneps formula and got the same BAD DEVICE ID, so that's highly curious and I suppose could suggest either a bad cable, a bad RaSCSI or something up with my Pi?? Fantastic to have a new clue!

@chickeneps I assume not but is there anything special you did to create the .hda? I used the web interface, chose .hda, 2000MiB, Masquerade as None, Format as None.

lukexi, 2022-11-11:
Gooooood gravy, I ordered a new shorter cable, it arrived today, and a 2GB disk formatted perfectly first try. I can't believe it was something so silly this whole time! 😵‍💫 Would probably be good to add a warning somewhere about potential symptoms with a bad cable so someone else can save a year of troubleshooting haha.

uweseimet, 2022-11-11:
@lukexi Good news! Does this mean this ticket can be closed.

Note that issues caused by bad cables or wrong termination are not that uncommon with SCSI. RaSCSI can be affected by it like any other peripheral.

lukexi, 2022-11-11:
I believe so — I'm just going to check the CD-ROM loading that's in the title of the issue and report back in a moment!

rdmark, 2022-11-11:
@lukexi If you have a moment to spare, please add a section to https://github.com/akuker/RASCSI/wiki/Troubleshooting

lukexi, 2022-11-11:
Already done : )

lukexi, 2022-11-11:
CD-ROM loads perfectly. I'm overjoyed, thanks everyone for all your help and comments!

closed 2022-11-11 by lukexi