PiSCSI/piscsi #847: develop branch won't build on aarch64

alicecrawford, 2022-09-19:
I was trying to build the develop branch to play around with the modified Apple CD code and couldn't get it to build.  A quick git bisect and I found that the architecture check in `config.h` was changed to check for not ARM instead of for x86:

https://github.com/akuker/RASCSI/blob/05db0e46881f6d18a02054f0c3aae8e9abc8bae7/src/raspberrypi/config.h#L23

The check here is incomplete, it should read something like: `#if !defined(__arm__) && !defined(__aarch64__)` instead to cover 32-bit and 64-bit ARM systems.  It may also be good to check that the OS isn't macos too; just in case someone is trying to build on an M1 Mac.

uweseimet, 2022-09-19:
@alicecrawford I guess this issue will be resolved after merging the feature_more_code_cleanup branch. Can you please check whether the feature_more_code_cleanup compiles in your environment?

alicecrawford, 2022-09-19:
That one does compile, but it does not include the breaking change.

uweseimet, 2022-09-20:
@alicecrawford Yes, I reverted that change because of other issues.

closed 2022-09-21 by uweseimet