PiSCSI/piscsi #860: Use the PI's MAC address as the DaynaPort's MAC address

uweseimet, 2022-09-24:
Currently there is a TODO because the DaynaPort emulation uses a hard-coded MAC address. IMO the only alternative to use a hard-coded address is the use the PI's MAC address.
Would there be any issues with this solution?

uweseimet, 2022-09-24:
@rdmark My impression is that you know much more about networking than I do. Do you think my suggsted solution would cause any issues?

rdmark, 2022-09-24:
@uweseimet I'm hardly an expert either, but wouldn't using the RPis MAC address as-is lead to conflicts when both the RPi and the computer using the DaynaPort emulation are connected to the same network?

There is one solution discussed in https://github.com/akuker/RASCSI/issues/598 where the MAC address is derived from the RPi's. This seems like a more sound approach to me.

uweseimet, 2022-09-24:
@rdmark I missed thi other ticket, thank you for pointing this out. I will close this ticket, because it is effectiveyl a duplicate.

closed 2022-09-24 by uweseimet