PiSCSI/piscsi #864: Support loading RX2/REX file using SMDI

Ruxton, 2022-09-26:
# Support loading RX2/REX file using SMDI

This is kind of a two-part feature request.

It'd be awesome to stick Recycle files (RX2/REX) on the device and load the individual samples into the sampler via SMDI.

https://github.com/ehasting/zoeos/tree/master/ExternalLibsAndTools/Deprecated/NativeLibs/OpenSMDI

rdmark, 2022-09-27:
@Ruxton Do you know of usage examples of OpenSMDI with other SCSI storage devices? It is not clear to me that any kind of work is needed on the RaSCSI side to enable this protocol.

A facility for uploading arbitrary file to the RPi's file system is easy enough, if that's what you're asking about.

Ruxton, 2022-09-27:
> A facility for uploading arbitrary file to the RPi's file system is easy enough, if that's what you're asking about.

This would be the first part, the second part would then need a "SMDI Send to device" on those files that:

1. processes the REX/RX2 into it's individual slices/samples
2. initate and SMDI session over SCSI
3. send the samples over SCSI using SMDI

Examples of things that do this:
* Recycle 2.0 allows you to send like this, but there is no longer support and getting it working these days is becoming annoying
* Chicken Systems Translator Professional can send packs of samples like this
* https://sourceforge.net/projects/esmdi/files/esmdi/0.1/
* SoundForge used to do this, not sure if it still does

Also, there is:
* [Peavey's original spec for SMDI]
(https://github.com/ehasting/zoeos/raw/master/ExternalLibsAndTools/Deprecated/NativeLibs/OpenSMDI/UnknowCrap/smdi.pdf)
* [OpenSMDI Manual](https://web.archive.org/web/20010611101847/http://www.teklab.com/~opensmdi/smdi.html#SMDIProtocol)

rdmark, 2022-09-29:
@Ruxton Understood, thanks for the overview. I had a look at the OpenSMDI library code and quickly realized that it is for Windows, and depends on VS2007 to compile. Are you aware of a Linux version of this library? In my opinion it is out of the scope for RaSCSI to implement a proprietary protocol and manage compatibility with a range of non-standard hardware clients, so we absolutely must have a 3rd party library that provides this layer to even consider the feature request.

While I can see how this would be an incredibly useful feature for sampler users, with the information we have right now I honestly don't think it's feasible.

rdmark, 2022-09-29:
I uploaded the original spec to archive.org for posterity and discoverability -- https://archive.org/details/smdi_20220929
It is a non-free file so I don't think it's correct to be distributing it in a GPL repo.

rdmark, 2022-09-29:
Set up forks for OpenSMDI and ESMDI on GitHub (distributed under GPL3) and started modernizing the C code to compile on Linux.
https://github.com/rdmark/OpenSMDI
https://github.com/rdmark/ESMDI

Still absolutely no guarantees, but I felt compelled to at least get these two pieces of code to a working state as a POC. Plus, it seems noone has ever properly put them under revision control, and were floating around in tarballs on archive.org and elsewhere, so I thought this was the dignified thing to do. :)

Ruxton, 2022-09-30:
thanks for your efforts @rdmark 

It's definitely something i could help with as a programmer and owner of an e-mu ultra sampler. I just dont have my RASCSI setup yet due to the Pi shortages, so it's gonna be a while til i come across one or they start shipping again O_O and with no exposure to this project, i felt it was best to raise the request first and get the conversation happening.

rdmark, 2022-09-30:
I got both OpenSMDI and ESMDI to a state where they compile and link on Debian (gcc 10 and 12), x86_64 and ARM. The statically linked esmdi crashes instantly, so there's certainly still some more progress to be made. But at least it is something!

If you're good at debugging Linux software please by all means clone the repos and see what you can figure out! PRs are very welcome. :)

BTW, I set it up so that OpenSMDI is a submodule of ESMDI. So you need to initialize the submodule when closing ESMDI. F.e.:
```
git clone --recurse-submodules [repo URL]
```

After cloning ESMDI, you want to first compile the OpenSMDI libraries (cd opensmdi && make), and ESMDI second.