PiSCSI/piscsi #869: Make rascsi.cpp more granular

uweseimet, 2022-09-27:
rascsi.cpp is too big and deals with a lot of unrelated things. The code shall be split into specialized classes, which can at least partially be tested with unit tests. Candidates are:

- RascsiService (thread handling)
- RascsiExecutor (command execution)

uweseimet, 2022-09-29:
@rdmark These and other changes are available in the feature_more_unit_tests branch, which is based on the feature_memory_management branch. For the latter there is already a PR, which I hope will be reviewed soon. Right after feature_memory_management has been approved I intend to create a PR for feature_more_unit_tests.
feature_more_unit_tests provides unit tests not only for SCSI command implementations, but also for the rascsi/rasctl commands, e.g. for detaching or attaching devices. I have tested this branch with buster and bullseye. If you have some time left it would be great if you could also run some tests with your setup.

rdmark, 2022-09-29:
@uweseimet Let me test your branch with my Macs today. I'll let you know how it goes. Immediately on runtime there was a warning with the latest code about invalid log levels that I haven't seen before:
```$ sudo ./rascsi
SCSI Target Emulator RaSCSI Reloaded version 22.09 --DEVELOPMENT BUILD-- (Sep 29 2022, 10:41:30)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2022 Contributors to the RaSCSI Reloaded project
Connect type: FULLSPEC
[2022-09-29 11:44:31.101] [info] Set log level to 'info'
[2022-09-29 11:44:31.105] [warning] Invalid log level ''
[2022-09-29 11:44:31.106] [info] No devices currently attached.
No devices currently attached.
```

uweseimet, 2022-09-29:
@rdmark Thank you for testing. Can you please provide the command line that resulted in the log level warning?
Foy any issue you report I will try to add a unit test.

rdmark, 2022-09-29:
@uweseimet I literally just started rascsi without any parameters. No other clients running that could be sending invalid protobuf commands.

uweseimet, 2022-09-29:
@rdmark Yes, I only noticed that after writing my initial comment. Usually I start rascsi with parameters and the first lines of output have scrolled away, this is why I did not notice this message myself.

uweseimet, 2022-09-29:
@rdmark I fixed the log level issue, but while testing more I stumbled upon a memory corruption issue. I suggest that you do not test any further until I have identified and fixed the problem.

uweseimet, 2022-09-30:
@rdmark Looks as if I have fixed the memory corruption issue. The valgrind tool (memory issue check for C/C++) was very helpful once more.

uweseimet, 2022-09-30:
@rdmark I could easily reproduce this issue and with the latest feature_more_unit_tests branch it should be fixed. In case of issues with sending interface commands trace logs usually don't help much. They are in particular important when it is about issues on the SCSI command/protocol level.

By the way, when there are issues with interface commands it does not matter how long rascsi has already been running. The code for executing commands is independant of the code that deals with the SCSI command handling. The command handling happens in a separate thread, which is waiting for commands to arrive on the socket used for the protobuf interface. As long as this thread does not crash (like in this case) the rest of rascsi simply keeps running.

uweseimet, 2022-10-01:
@rdmark Can you please create a new SonarCloud C++ profile (an inherited one) and disable the rule that says that "using namespace" should not be used in header files? For rascsi this rule is not relevant, but it is reported for most header files. I can suppress it in the code but I would like to avoid suppressing messages in the code.

uweseimet, 2022-10-01:
@rdmark I updated the branch with a more transparent command validation, which up to now was essentially based on legacy code.
In case there are still issues with detaching/inserting etc. one can see quite easily (at least I hope so) how the validation and execution works in this method in rascsi_executor.cpp:

```
bool RascsiExecutor::ProcessCmd(const CommandContext& context,
  const PbDeviceDefinition& pb_device, const PbCommand& command, bool dryRun)
```

rdmark, 2022-10-01:
@uweseimet I updated the C++ Sonar profile to turn off those two code smells. Hopefully less noisy reports moving forward. As for the exclusion filters, the only path that we explicitly exclude right now is src/raspberrypi/tests/* so at a glance I cannot say where the problem lies. Might require a bit more investigation.

uweseimet, 2022-10-01:
@rdmark Thank you. Can you please remove the exclusion for the /tests/ folder? I would like to add this exclusion to the github action, because it can be configured externally. But I might not even want to exclude this folder at all, except for the code coverage.
In general I would like to keep the SonarCloud configuration external as far as possible. (For the two code smells there is no external alternative.)

uweseimet, 2022-10-01:
@rdmark Regarding the files where I would have expected problems, I may simple have been wrong. There definitely are code issues, but probably none SonarCloud would report.

rdmark, 2022-10-01:
> @rdmark Thank you. Can you please remove the exclusion for the /tests/ folder? I would like to add this exclusion to the github action, because it can be configured externally. But I might not even [...]

Is it possible to do that kind of granular exclusion from test coverage in the github action config?
And, you're thinking that running static analysis on the test code can also produce valuable feedback?

uweseimet, 2022-10-01:
@rdmark Yes, you can configure the code in the action with sufficient granulariy. Actually, the setting is the same (same arguments) as in the UI.
Regarding test code analysis, this can be valuable, because it can avoid errors in the test code that might result in a test making a wrong assessment. One can turn it off in the github action any time anyway.

uweseimet, 2022-10-01:
@rdmark By the way, even coverage data can sometimes be useful for tests: If the coverage is not 100% not all tests have been executed, which indicates a problem with the test setup. In the past I already stumbled upon such an issue with Java unit tests not being executed.

rdmark, 2022-10-01:
@uweseimet That makes sense. I deactivated the exclusion rule in Sonar. FYI, we still have a code duplication exclusion in place for src/raspberrypi/test/* (Patterns used to exclude some source files from the duplication detection mechanism.)

uweseimet, 2022-10-01:
@rdmark Thank you. Can you please also disable the duplication exclusion? It can also be configured in the action, and I would also like to see what's reported here.

rdmark, 2022-10-01:
@uweseimet Done

uweseimet, 2022-10-01:
@rdmark Thank you!

uweseimet, 2022-10-04:
@rdmark Regarding my expectation that SonarCloud should still report issues for the old code I think I know the resolution: For branches only issues for new (edited) code are reported. Per default only for the master branch there is a full report. This means that only after the next release (after merging into master) one can see the actual status.
IMO we would profit from having a full analysis also for the develop branch. Can you please make the develop branch a long-living branch? See https://community.sonarsource.com/t/how-to-create-long-living-branches-on-sonarcloud/11386.

uweseimet, 2022-10-04:
@rdmark I just tested this by giving a branch a name starting with "branch-", and indeed more issues are reported than on a non long-living branch with the same code. No issues where I would have expected them, though ;-).
But I noticed that there are still issues in the css/html code which are not related to being compatible with old browsers.

closed 2022-10-04 by uweseimet