PiSCSI/piscsi #870: Create sub-folders for rascsi/rasctl-specific classes

uweseimet, 2022-09-27:
rascsi/rasctl-specific classes should be moved to new sub-folders 'rascsi' and 'rasctl'.

uweseimet, 2022-10-02:
@akuker @rdmark If there are not objections to this change I will start with it after https://github.com/akuker/RASCSI/pull/878 has been merged.

rdmark, 2022-10-02:
I have no concerns. It sounds like a sound approach.

uweseimet, 2022-10-03:
The feature_sub_folders branch (based on feature_more_unit_tests) is the result of creating sub-folders. rascsi.cpp and rasctl.cpp stay outside for reasons of consistency, and because otherwise the main() method of rascsi.cpp would collide with the main() method of the unit tests.
As part of this change I moved some code from command_util.cpp to command_context.cpp, because this cleans up dependencies. Now rasctl.cpp does not any longer depend on the localizer. Before there was an indirect dependency.