PiSCSI/piscsi #873: powerview attach failure on raspberry pi zero 2

stevebourg, 2022-09-28:
# Info

- Which version of Pi are you using: 
  - pi@raspberrypi:~/RASCSI $ grep Model /proc/cpuinfo 
    Model		: Raspberry Pi Zero 2 W Rev 1.0
  - pi@raspberrypi:~/RASCSI $ cat /etc/debian_version 
    10.13
  - pi@raspberrypi:~/RASCSI $ uname -a
    Linux raspberrypi 5.10.103-v7+ #1529 SMP Tue Mar 8 12:21:37 GMT 2022 armv7l GNU/Linux
- Which github revision of software:
  - (9/27/22) git checkout feature_powerview
  - (9/27/22) git pull
  - pi@raspberrypi:~/RASCSI $ rasctl -V
    rascsi server version: 22.1.1
- Which board version: 
  - pi@raspberrypi:~/RASCSI $ grep Model /proc/cpuinfo 
    Model		: Raspberry Pi Zero 2 W Rev 1.0
  - https://samplerzone.com/products/chicken-rascsi-sd-scsi-drive
- Which computer is the RaSCSI connected to: 
  - Mac SE/30

# Describe the issue
git checkout feature_powerview
git pull
./easyinstall.sh
..
  2) install or update RaSCSI Service

pi@raspberrypi:~/RASCSI $ rasctl -i 5 -c attach -f powerview
Error: Invalid protobuf header (size) recevied

pi@raspberrypi:~/RASCSI $ rasctl -l
No images currently attached.

Damien130, 2022-09-29:
> I've seen exactly the very same behavior, I was about to try again, but I can't even compile it in bullseye: `g++ -O3 -Wall -Werror -DNDEBUG -Wno-psabi -std=c++17 -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP -DFMT_HEADER_ONLY -DCONNECT_TYPE_FULLSPEC -c ./devices/scsi_powerview.cpp -o obj/fullspec/scsi_powerview.o ./devices/scsi_powerview.cpp: In member function ‘virtual bool SCSIPowerView::Init(const std::map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >&)’: ./devices/scsi_powerview.cpp:404:6: error: cast from ‘char*’ to ‘int’ loses precision [-fpermissive] 404 | if ((int)fb == -1){ | ^~~~~~~ In file included from ./devices/disk.h:20, from ./devices/scsi_powerview.h:23, from ./devices/scsi_powerview.cpp:21: ./devices/scsi_powerview.cpp: In member function ‘bool SCSIPowerView::WriteUnknownCC(const DWORD*, const BYTE*, DWORD)’: ./devices/scsi_powerview.cpp:491:12: error: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘long unsigned int’ [-Werror=format=] 491 | LOGERROR("%s Received Unknown CC data that is larger (%d bytes) than allocated size (%d bytes)", __PRETTY_FUNCTION__, length, sizeof(unknown_cc_data)); | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~~~~~~~~~~~~~~~~ | | | long unsigned int ./log.h:22:35: note: in definition of macro ‘SPDLOGWRAPPER’ 22 | snprintf(logbuf, sizeof(logbuf), __VA_ARGS__); \ | ^~~~~~~~~~~ ./devices/scsi_powerview.cpp:491:3: note: in expansion of macro ‘LOGERROR’ 491 | LOGERROR("%s Received Unknown CC data that is larger (%d bytes) than allocated size (%d bytes)", __PRETTY_FUNCTION__, length, sizeof(unknown_cc_data)); | ^~~~~~~~ ./devices/scsi_powerview.cpp:491:89: note: format string is defined here 491 | LOGERROR("%s Received Unknown CC data that is larger (%d bytes) than allocated size (%d bytes)", __PRETTY_FUNCTION__, length, sizeof(unknown_cc_data)); | ~^ | | | int | %ld In file included from ./devices/disk.h:20, from ./devices/scsi_powerview.h:23, from ./devices/scsi_powerview.cpp:21: ./devices/scsi_powerview.cpp: In member function ‘bool SCSIPowerView::WriteColorPalette(const DWORD*, const BYTE*, DWORD)’: ./devices/scsi_powerview.cpp:511:12: error: format ‘%d’ expects argument of type ‘int’, but argument 6 has type ‘long unsigned int’ [-Werror=format=] 511 | LOGERROR("%s Received Color Palette that is larger (%d bytes) than allocated size (%d bytes)", __PRETTY_FUNCTION__, length, sizeof(color_palette)); | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ ~~~~~~~~~~~~~~~~~~~~~ | | | long unsigned int ./log.h:22:35: note: in definition of macro ‘SPDLOGWRAPPER’ 22 | snprintf(logbuf, sizeof(logbuf), __VA_ARGS__); \ | ^~~~~~~~~~~ ./devices/scsi_powerview.cpp:511:3: note: in expansion of macro ‘LOGERROR’ 511 | LOGERROR("%s Received Color Palette that is larger (%d bytes) than allocated size (%d bytes)", __PRETTY_FUNCTION__, length, sizeof(color_palette)); | ^~~~~~~~ ./devices/scsi_powerview.cpp:511:87: note: format string is defined here 511 | LOGERROR("%s Received Color Palette that is larger (%d bytes) than allocated size (%d bytes)", __PRETTY_FUNCTION__, length, sizeof(color_palette)); | ~^ | | | int | %ld cc1plus: all warnings being treated as errors make: *** [Makefile:167: obj/fullspec/scsi_powerview.o] Error 1 pi@raspberrypi:~/RASCSI $ `

I was encountering this issue. Do not use 64-bit Bullseye. 32-bit compiles without error.

dramirez1972, 2022-10-01:
Thanks Damien, installed on 32 bit bullseye, but now I have the same problem (kind of) as Steve:

`pi@raspberrypi:~/RASCSI $ rasctl -c attach -i 4 -f powerview
Error: Initialization of  device, ID 4, unit 0 failed
pi@raspberrypi:~/RASCSI $ rasctl -i 6 -c attach -t scdp -f wlan0:10.10.20.1/24
pi@raspberrypi:~/RASCSI $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
pi@raspberrypi:~/RASCSI $ rasctl -c attach -i 4 -f powerview
Error: Invalid protobuf header (size) recevied
pi@raspberrypi:~/RASCSI $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
pi@raspberrypi:~/RASCSI $ rasctl -c attach -i 4 -f powerview
Error: Initialization of SCPV device, ID 4, unit 0 failed
pi@raspberrypi:~/RASCSI $ rasctl -c attach -i 4 -f powerview
Error: Initialization of SCPV device, ID 4, unit 0 failed
pi@raspberrypi:~/RASCSI $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
pi@raspberrypi:~/RASCSI $
`

akuker, 2022-10-01:
Just a note that the PowerView functionality is in `alpha` state. But, I appreciate the feedback!

Since this is an `alpha` feature, I'm going to tag this as an enhancement. I'd like to leave it open so I remember to look into it soon.

dramirez1972, 2022-10-05:
Thanks @akuker @uweseimet, what a wonderful product the RaSCSI is, besides mounting hard drives and ISO images, being able to bridge Ethernet to my SE/30, powerview will be just amazing!

BTW: I have two raspberrypi 3's, one with a clean Bullseye install (the one powerview doesn't work), and another also running Bullseye but was upgraded from Stretch to Buster, then Bullseye and powerview does indeed attach in that one!

pi@raspberrypi2:~/images $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  1 |   0 | SCHD | /home/pi/images/600MB.hda
|  2 |   0 | SCHD | /home/pi/images/1900MB.hda
|  4 |   0 | SCPV | Radius PowerView
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
pi@raspberrypi2:~/images $ rasctl -v
rasctl version: 22.01.1
pi@raspberrypi2:~/images $ uname -a
Linux raspberrypi2 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
pi@raspberrypi2:~/images $

pi@raspberrypi:~ $ rasctl -c attach -i 4 -f powerview
Error: Initialization of SCPV device, ID 4, unit 0 failed
pi@raspberrypi:~ $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  1 |   0 | SCHD | /home/pi/images/600MB.hda
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
pi@raspberrypi:~ $ rasctl -v
rasctl version: 22.01.1
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.61-v7+ #1579 SMP Fri Aug 26 11:10:59 BST 2022 armv7l GNU/Linux
pi@raspberrypi:~ $

Thanks for your help!

dramirez1972, 2022-10-31:
Hi guys, I just noticed in the rascsi.log:

Oct 30 18:04:39 raspberrypi RASCSI[479]: [2022-10-30 18:04:39.353] [warning] Unable to open /dev/fb0. Do you have a monitor connected?
Oct 30 18:04:39 raspberrypi RASCSI[479]: [2022-10-30 18:04:39.353] [warning] void SCSIPowerView::fbcon_cursor(bool) Unable to open /dev/tty1
Oct 30 18:04:39 raspberrypi RASCSI[479]: [2022-10-30 18:04:39.353] [error] Initialization of  device, ID 4, unit 0 failed

Connecting a monitor resolved my issue!

pi@raspberrypi:~ $ rasctl -c attach -i 4 -f powerview
pi@raspberrypi:~ $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  4 |   0 | SCPV | Radius PowerView
+----+-----+------+-------------------------------------
pi@raspberrypi:~ $

MEGAHeinz, 2022-10-31:
Did you have success after that? I can see the GPU with scsi probe, but when i want to install the driver, it says no compatible hardware found

stevebourg, 2022-11-11:
An update.  I found the following in /var/log/rascsi.log

Sep 27 21:16:44 raspberrypi RASCSI[428]: [2022-09-27 21:16:44.163] [error] PowerView emulation only supports 16 bits per pixel framebuffer. Currently set to 32
Sep 27 21:16:44 raspberrypi RASCSI[428]: [2022-09-27 21:16:44.163] [error] Try running "fbset -depth 16"

Having the monitor attached and running 'fbset -depth 16' command, I was then able to attach the powerview device at the cli.  Using the PowerView 1.01 driver, I was able to get a working (color, 800x600) 2nd monitor on my SE/30 (System Software 7.5.5).  I had some thrash between trying to get the 1.01 and the 2.1 driver to work - you might have to manually copy the PowerView files into System Folder / Control Panels.  The overall UI is slower on the PowerView display (which I've read is to be expected).  Right now I get a 800x600 Macintosh in the upper left corner of my much larger / higher resolution display.  I'm hoping there's an option to scale it.

akuker, 2022-11-11:
> I'm hoping there's an option to scale it.

Not currently. There will be in the future though!

akuker, 2023-10-14:
None of the other devs seem interested in this feature. I will not be able to finish this functionality. If anyone is willing/able to pick it up, I'm happy to provide them with a PowerView at a reasonable cost. 

I'm going to close this ticket since this feature didn't ever make it into the mainline software.

closed 2023-10-14 by akuker