PiSCSI/piscsi #876: What are the purposes of the two LEDs on GPIO 4/5

lukecyca, 2022-10-01:
I am using hardware 2p4a:

GPIO 4 is labeled `PI-ACT` in the schematic and GPIO 5 is labeled `DBG_LED`. On the PCB silkscreen they are labeled `ACT` and `ENB` respectively.

I don't see ACT flash or anything to indicate activity on the SCSI bus, as I expected. Under my regular usage it seems like ENB is lit basically all the time that the `rascsi` software is running. If I stop the service, it shuts off, and ACT becomes dimly lit.

Can anyone clarify the purpose of these LEDs, or where in the codebase they are toggled? Ultimately I am wanting to wire some external LEDs on an enclosure I am building, but I first want to understand what they do. Thanks!

rdmark, 2022-10-01:
@lukecyca Does this answer your question? https://github.com/akuker/RASCSI/wiki/Connecting-the-RaSCSI#what-are-all-these-green-lights

lukecyca, 2022-10-01:
It absolutely does. Thank you! I'll see myself out now... 🤦‍♂️

closed 2022-10-01 by lukecyca

rdmark, 2022-10-01:
Great! :)

lukecyca, 2022-10-01:
For some reason my activity light wasn't working. Some troubleshooting using `raspi-gpio get` showed that GPIO 4 was in INPUT mode. I set it to OUTPUT mode and now it works as expected.

It seems that `rascsi` might be assuming these pins are in output mode, rather than setting them so during startup.