PiSCSI/piscsi #879: Template for displaying manpages in the Web UI

rdmark, 2022-10-01:
Since we have a proven pattern for displaying terminal output in a Flask template, I think it'd be handy to show rascsi manpages. In particularly helpful for folks who aren't comfortable with the command line

uweseimet, 2022-10-01:
@rdmark Just a personal opinion from someone who regularly uses the command line: If I am willing to use the command line, I am also willing to run 'man' in order to display the manpage in my console. Using a web UI for this while I am using the console anyway is IMHO not that helpful.

rdmark, 2022-10-01:
@uweseimet Agreed that the rasctl manpage is not very useful if you don't use the command line. The main current use case that I see is that the rascsi manpage contains documentation about supported file extensions and what they're used for. Instead of maintaining effectively duplicated help text in the Web UI I want to refer to the manpage.

rdmark, 2022-10-02:
Here's my take on this https://github.com/akuker/RASCSI/pull/881
It reduces the maintenance overhead of the Web UI

uweseimet, 2022-10-02:
@rdmark I see. A better user experience might be to generate HTML from the raw manpage data. See https://linux.die.net/man/1/man2html. Not sure whether this is better, though.

rdmark, 2022-10-03:
@uweseimet Thanks for the lead with man2html. I experimented a bit with it. It general, it creates pretty good html. The challenge will be to use the output in the context of the Flask app. We need to render it inside a Flask template, so at the very least we need heuristics to strip the html/title/body tags, as well as the content header and footer.

As an MVP, I went ahead and used the contents of the generated txt manpage inside a <pre> tag. This works pretty well in modern browsers.

Let me keep experimenting with man2html and reopen this ticket if I find a better solution!

closed 2022-10-03 by rdmark

rdmark, 2022-10-04:
It was actually fairly easy to get man2html to output what we needed. :)

https://github.com/akuker/RASCSI/pull/887