PiSCSI/piscsi #891: CD-ROM emulation freezes Performa 6200

MrGasS, 2022-10-04:
# Info

- Which version of Pi are you using: Raspberry Pi Zero W
- Which github revision of software: both v22.02.01 and v22.08.01
- Which board version: custom RaSCSI board (my RaSCSI Reloaded THT PCB)
- Which computer is the RaSCSI connected to: Apple Macintosh Performa 6200/75/CD

# Describe the issue

Hi!

I just want to leave this problem here, as I noted that RaSCSI's CD-ROM emulation freezes my Macintosh Performa 6200.
Sometimes it freezes during boot, sometimes it does that after I open an application from the ISO image.
Emulating an Apple 600e or a Sony drive doesn't change the situation.

I want to clarify that I'm using my custom RaSCSI board with MC74F245N transreceiver: despite this, CD emulation works just fine in every other Mac that I have, examples are the IIci, IIsi, Quadra 700, G3/266 (also boots fine from ISO) and G4/800 QS /w Adaptec 2906.
However, I still need to give a try on my 6500/275, but I need to get it out from storage first.

Every ISO image works perfectly on the Macs I already listed and they works fine on this Performa 6200 with a BlueSCSI on firmware v1.1-20220917.

I would love if a Performa 6200 user could give me a feedback about this issue.

Thank you.

MrGasS, 2022-10-04:
Okay, I must add a detail: few minutes ago, the 6200 froze with the BlueSCSI, so, there's something wrong with the 6200 in general, my 6200 or its SCSI controller.

However, my Quadra 700 works fine with CD emulation from both RaSCSI and BlueSCSI.

Again, any feedback from 6200's users are appreciated.

uweseimet, 2022-10-10:
@MrGasS I recommend that you ask for feedback in the Discord channel or in the github Discussions area.
As it turned out that your issue is not caused by rascsi, i.e. this is not a rascsi development issue, I will close this ticket.

closed 2022-10-10 by uweseimet