PiSCSI/piscsi #911: Use C++ filesystem library for image file operations

uweseimet, 2022-10-12:
Instead of the low-level C API the C++ filesystem library shall be used for image file operations.

akuker, 2022-10-19:
As a request - however this is implemented, I'd like to have a path to support FreeRTOS (or something similar) file system APIs. I think there is an opportunity to run RaSCSI on FreeRTOS.

Would you consider keeping this as a separate layer (rather than calling the C++ FS APIs directly) so that we can swap OS's in the future?

uweseimet, 2022-10-19:
@akuker Can you please make this a separate ticket? I have alaready completed the changes and unit tests required for this ticket.

akuker, 2022-10-22:
I'm fine doing it under a different ticket, since you've got the changes already done. I'm sure there will be other changes to port to FreeRTOS, so we can cross that bridge when we get there.

closed 2022-10-25 by uweseimet