PiSCSI/piscsi #913: Making images directory visible to netatalk

ctb68, 2022-10-14:
# Info

- Which version of Pi are you using:
- Which github revision of software:
- Which board version:
- Which computer is the RaSCSI connected to:

# Describe the issue
It would be a nice touch to make the RaSCSI images directory appear as an AppleShare volume if netatalk is installed via easyinstall.sh. This would allow someone with a modern Mac to install images easily via the Finder without needing to "upload" from the web client.

rdmark, 2022-10-15:
I can see how this can be a useful default configuration for users who have modern Macs. The only consideration is that Netatalk creates metadata in directories that it shares:

```
drwxr-xr-x  2 root  root        4096 Oct 15 10:34  .AppleDB
drwxrwxr-x  2 root  dmark       4096 Oct 15 10:34  .AppleDesktop
drwxrwxr-x  2 dmark dmark       4096 Oct 15 10:34  .AppleDouble
```

@uweseimet RaSCSI will ignore dot files/directories when it scans the images dir, won't it?

uweseimet, 2022-10-15:
@rdmark Yes, files and folders starting with a dot are ignored.

By the way, this ticket should probably be extended to make image directories visible via samba (and potentially nfs), because there are also non-Mac users.

uweseimet, 2022-10-15:
Just for the record: Initially this sounded like a good idea to me, but after thinking more about it I think it should not be done (at least not by default) because of security issues. If a users wants data to be exported this is fine as long as the user is aware of the fact that data may potentially be made accessible to others.
In a world where data security is becoming more important all the time, regardless on which system, features like this should be considered very carefully. What would happen if any software installation was exporting parts of your data? Well, maybe that's actually the case, you never know ;-).

rdmark, 2022-10-15:
@uweseimet I agree completely that we should be mindful about data security. In the PR, I've flagged this option as "experimental" as an extra precaution. https://github.com/akuker/RASCSI/pull/916

The reason that we spend this extra effort providing configuration facilities for Netatalk 2 in RaSCSI (as opposed to Samba), is because it's been gone from Linux package repos for at least half a decade, and in a broken state for Linux for about the same amount of time.

Just for the record, I believe Mac OS X supports Samba out of the box starting with 10.9 Mavericks, in 2013. The benefit with Netatalk 2 is that we get compatibility all the way back to the earliest networked Mac OS versions.

rdmark, 2022-10-18:
Added as a stand-alone, experimental feature. The primary Netatalk installation script was not changed.

closed 2022-10-18 by rdmark