PiSCSI/piscsi #922: Request: Ability to copy files directly to an attached drive via web interface

durosity, 2022-10-20:
# Info

- Which version of Pi are you using: RP 4B
- Which github revision of software: 22.8.1 [b5ddb88]
- Which board version: 2.3
- Which computer is the RaSCSI connected to: Quadra 950

# Describe the issue
Unless I'm missing something blatantly obvious, it'd be amazing if there was a way to download files (or drag to from a local machine) an area on the web config page and have those files copied directly onto one of the attached drives mounted on the RASCSI device, much like when copying files for use with AppleShare, but much easier to access when you're using a setup that doesn't allow for AppleTalk to work.

akuker, 2022-10-22:
Hi @durosity !

Thank you for the suggestion. This would be a rather difficult undertaking (at least for me). The MacOS doesn't expect the SCSI drive to be changing underneath it. You'd end up with corrupted drive images if you just added files to the drive while its mounted by MacOS (or any other operating system). 

The host controller interface that Gimons developed for the X68000 did something similar to what you're wanting to do , but it required a special host (Sharp X68k) driver in order to make it work. 

We want to update RaSCSI to work with the BlueSCSI transfer tool (https://github.com/erichelgeson/BlueSCSI-Toolbox#bluescsi-transfer-utility), which would also allow you to transfer files to/from the Raspberry Pi to your Mac. However, that functionality isn't pulled into RaSCSI yet. 

Right now, the easiest way to achieve what you're looking to do is to create a Netatalk share on your Raspberry Pi and share it with your Macintosh over Ethernet (potentially using the DaynaPort SCSI Ethernet adapter functionality)

rdmark, 2022-11-05:
@durosity You also have the "Download to ISO" functionality in the Web UI. If you have an "empty" CD-ROM drive attached you can get the iso image immediately inserted and mounted by Mac OS.

Beyond that, as akuker say directly writing to a hard drive image that is in use might lead to data corruption so it is not an advisable solution. I will close this ticket as WONTFIX.

closed 2022-11-05 by rdmark