PiSCSI/piscsi #927: Allow easyinstall.sh to run without user input

akuker, 2022-10-22:
# Info

- Which version of Pi are you using: N/A
- Which github revision of software: v22.10.01
- Which board version: N/A
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

When building the Raspberry Pi OS release images, there is one step that requires user interaction half-way through. The easyinstall.sh script will ask for the user password to support the sudo commands. It would be useful if the easyinstall script could be run 100% without user input. We could potentially get to a point where Github actions can create a Raspberry Pi OS disk image for each commit to develop.

rdmark, 2022-10-22:
I ran through the latest develop version of the script, and the only place where it explicitly paused was in ```installRaScsiCtrlBoard()``` which I don't think is relevant for the build script, since it shouldn't get to this function (unless I'm mistaken!)

nucleogenic, 2022-10-24:
Just as a heads up, there's a few examples where I've used the `--headless` param (`$HEADLESS`) to instruct the script to use defaults/skip confirms, for non-interactive activity.

akuker, 2022-11-01:
The user prompt that I run into is the `sudo -v` where it prompts for the user's password. 

Here's the portion of the OS creation script that calls easyinstall.sh. Maybe I just need to add `--headless`?

akuker, 2022-11-02:
[image]

rdmark, 2022-11-02:
@akuker Is it feasible to set the NOPASSWD flag for the user in your environment? This would entirely bypass the ```sudo -v``` command.

rdmark, 2022-11-02:
@akuker Can you see if this does the trick for you? https://github.com/akuker/RASCSI/pull/963

closed 2022-11-04 by rdmark