PiSCSI/piscsi #955: Improve testability of rascsi.cpp and rasctl.cpp

uweseimet, 2022-11-01:
In order to improve the testability of rascsi.cpp and rasctl.cpp all static code except for main() should be moved into a respective class. The methods of this class can then be tested/mocked.
The result is essentially the same as what you typically have when using SpringBoot: A static main() method and the rest is encapsualted in classes.

uweseimet, 2022-11-02:
@rdmark You may remember that I was wondering why SonarCloud does not report any issues for rascsi.cpp, rasctl.cpp, scsimon.cpp and rasdump.cpp, even though when looking at the code I can easily identify issues that should be reported.

After starting to work on this ticket and moving most of the rasctl.cpp code into a separate class, SonarCloud reports issues for this code, just as expected. Are there any SonarCloud settings/exclusions that prevent files like the ones I mentioned above from being analyzed? As far as I can tell all files that have a main() class are not analyzed.

closed 2022-11-02 by uweseimet

rdmark, 2023-04-11:
> You may remember that I was wondering why SonarCloud does not report any issues for rascsi.cpp, rasctl.cpp, scsimon.cpp and rasdump.cpp, even though when looking at the code I can easily identify issues that should be reported.

Late response, but looking at those source files, they're all just 7 lines of code that imports headers, sets up some parameters and then return with a function call. I can see Sonar reporting issues against the code that they call. Maybe something changed since you asked this question? Anyhow, there are no analysis restrictions in the Sonar project that I can see..

uweseimet, 2023-04-11:
There was more code in these files before, but I moved this code to other files (the ones in the folders) when cleaning up the code, making it more granular, and in ordert to make it testable. But before the code in these files was not analyzed by Sonar. Only after I moved the code to separate files in the respective folders Sonar reported issues with this code.