PiSCSI/piscsi #956: Initializing a disk with HD SC Setup no longer works

rdmark, 2022-11-01:
In the latest develop:

When attempting to initialize a disk image with Apple HD SC Setup on a Mac Plus or SE (tested so far), the process leads to an error in HD SC Setup. This used to work in past RaSCSI software versions. I have yet to track down the breakpoint.

Tested with HD SC Setup 1.4 and 2.0

Using this inquiry data to spoof a drive that the tool recognizes:
```
MINSCRIB
M8425 - SCSI
209A
```

The tail of the trace log:
```
Nov 01 11:30:01 rasp32 RASCSI[27810]: [2022-11-01 11:30:01.695] [trace] void ScsiController::Receive() Phase: dataout
Nov 01 11:30:01 rasp32 RASCSI[27810]: [2022-11-01 11:30:01.695] [debug] Error status: Sense Key $05, ASC $26
Nov 01 11:30:01 rasp32 RASCSI[27810]: [2022-11-01 11:30:01.695] [trace] virtual void ScsiController::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
Nov 01 11:30:01 rasp32 RASCSI[27810]: [2022-11-01 11:30:01.695] [trace] virtual void ScsiController::Status() Status Phase, status is $02
Nov 01 11:30:01 rasp32 RASCSI[27810]: [2022-11-01 11:30:01.696] [trace] virtual void ScsiController::BusFree() Bus free phase

```

uweseimet, 2022-11-01:
@rdmark According to the log this is an issue with the MODE SELECT parameter list. Should not be a big deal to find out what's wrong and to fix it. But first I would like to wait until my pending changes (in two branches) have been merged. Hope that's OK for you.

This is most likely what causes the issue:
```
	if (!(cdb[1] & 0x10)) {
		// Vendor-specific parameters (SCSI-1) are not supported
		throw scsi_exception(sense_key::ILLEGAL_REQUEST, asc::INVALID_FIELD_IN_PARAMETER_LIST);
	}
```
The offending command tries to use vendor-specific parameters, which we cannot support because we don't know what the vendor-specific stuff would be. The code before v22.10.01 just ignored this and did nothing. I'm fine with being lenient again and doing nothing. At least now we know which software requires this behavior, so that an appropriate comment can be added.
The Atari software I tested this change with does not try to use vendor-specific extensions and thus the rest of the code is properly executed, by the way.

Maybe you can already test this change? You would just have to change the code snippet above (in scsi_command_util.cpp) to
```
	if (!(cdb[1] & 0x10)) {
            return;
	}
```

uweseimet, 2022-11-01:
@rdmark After looking at the current code again I noticed that the SCSI error code returned is wrong. It should be
```
	if (!(cdb[1] & 0x10)) {
		// Vendor-specific parameters (SCSI-1) are not supported
		throw scsi_exception(sense_key::ILLEGAL_REQUEST, asc::INVALID_FIELD_IN_CDB);
	}
```
In case you want to test this, before just add a "return" statement, can you please check whether using the correct error code, i.e. INVALID_FIELD_IN_CDB, also works?

rdmark, 2022-11-01:
@uweseimet Changing the error code to ```INVALID_FIELD_IN_CDB``` still leads to an error in HD SC Setup unfortunately. However, doing nothing (just the 'return') does allow the tool to complete the initialization of the drive! So your theory that the vendor specific parameters lead to the error seems to have been spot on.

uweseimet, 2022-11-01:
@rdmark OK, thank you for testing. I will add the required change to my pending changes in the testing_lambda branch.

uweseimet, 2022-11-01:
Resolved in the branch that also deals with https://github.com/akuker/RASCSI/issues/953.

uweseimet, 2022-11-02:
Branch name (for the upcoming PR) is feature-dispatcher-using-lambdas.

closed 2022-11-02 by uweseimet