PiSCSI/piscsi #969: rasdump and monitor folders are not analyzed by SonarCloud

uweseimet, 2022-11-05:
The files in the rasdump and monitor folders are not analyzed by SonarCloud. It turned out that they are only analyzed when they are part of the binary that is executed fo collect coverage information. This is not obvious from the SonarCloud documentation.
In order to resolve this issue the files in the respective folders have to be added to the unit test binary. In order to do that global variables have to be removed, because they prevent linking the binary because of symbol duplication issues.
In addition, the code needs to be cleaned up and SonarCloud issues need to be fixed.

uweseimet, 2022-11-05:
@akuker Note that rasdump (develop branch) segfaults during the bus initialization.

akuker, 2022-11-06:
Thank you for letting me know, @uweseimet  - that's not surprising. They are both going to need quite a few updates with the bpi changes.

uweseimet, 2022-11-06:
@akuker That's surprising, at least as far as rasdump is concerned, because essentially it just uses the BUS interface for everything, and this interface has not changed as far as I can tell.
Anyway, I have already spent considerable time on scsimon and rasdump in order to fix the numerous code issues. The next step would be a PR, as soon as https://github.com/akuker/RASCSI/pull/966 has been merged. I suggest to proceed exactly like that, i.e.that I will raise the PR right after the merge of  https://github.com/akuker/RASCSI/pull/966- Any required changes to scsimon and rasdump can be applied later, on the updated code.
Speaking of my pending PR, it would be great if you were reviewing it. There are not that many changes this time. Much less than my previous PRs ;-).

uweseimet, 2022-11-06:
@akuker Regarding rasdump and scsimon I could unblock myself by adding one line of code :). Note, though, that my remark on the bus interface is still valid.

uweseimet, 2022-11-06:
I could make some progress, but as far as I can tell initiator mode does not work anymore. It stopped working before my rasdump changes. I tested this with commit c41373d9.
```
>sudo ./bin/fullspec/rasdump -i 3 -b 7 -f dumpfile
RaSCSI hard disk dump utility version 22.11 --DEVELOPMENT BUILD-- (Nov  6 2022, 11:56:03)
[2022-11-06 12:02:21.856] [info] Detected device Raspberry Pi 4
[2022-11-06 12:02:21.857] [info] virtual bool GPIOBUS_Raspberry::Init(BUS::mode_e) I'm a pi 4
TARGET ID               : 3
BOARD ID                : 7
TEST UNIT READY ERROR -2
```
WIth the same hardware https://github.com/akuker/RASCSI/issues/131 shows the output when using release 22.10.01, where the INQUIRY was successful.

uweseimet, 2022-11-06:
There is yet another issue with the bus: When bus->Init() returns false (e.g. when you are not root), calling bus-Cleanup() (usually in a destructor) results in a segmentation fault. If you have a valid bus object (returned by the factory) calling a member function on it should not crash the process, at least not bus->Cleanup(), because you cannot really avoid this call if you want to clean up.

More suggestions that simplify creating a valid bus object:
```
#ifdef USE_SEL_EVENT_ENABLE
	SBC_Version::Init();
#endif
```
should be part of the factory code and should be executed when a client program calls "GPIOBUS_Factory::Create(). A client should not have to know about SBC_Version because the client in a bus object only.

In addtion, GPIOBUS_Factory::Create() should accept BUS::mode_e, so that a client does not have to call a sequence of several mandatory methods. Currently we need
```
SBC_Version::Init();
bus = GPIOBUS_Factory::Create();
bus->Init(BUS::mode_e::INITIATOR/TARGET);
```
This should only be
```
bus = GPIOBUS_Factory::Create(BUS::mode_e::INITIATOR/TARGET);
```
A factory in general is good for doing as much of the initialization work as possible. Otherwise clients need a lof of duplicate boilerplate code and are exposed to implementation details. If there is any issue, bus should be nullptr.

akuker, 2022-11-06:
> I could make some progress in the testing_analysis branch, but as far as I can tell initiator mode with rasdump does not work anymore. It stopped working before my rasdump changes. I tested this with commit [31dd063](https://github.com/akuker/RASCSI/commit/31dd063611e7fd6a6f3c08e806024d6939ed544c)

Rasdump didn't work when I tried it over a year ago either. I haven't heard of it successfully working for anyone with this fork.

akuker, 2022-11-06:
> There is yet another issue with the bus: When bus->Init() returns false (e.g. when you are not root), calling bus->Cleanup() (usually in a destructor) results in a segmentation fault. If you have a [...]

I don't disagree at all. This would be an easy change to make. Feel free to make it if you like.

TBH, I've really been struggling balancing time between work, life and RaSCSI. Trying to get this BananaPi functionality implemented and making the code able to handle different underlying hardware has been stressing me out a little too much. I have no objections if we just want to revert the GPIOBUS changes that I've done. 

Rasdump didn't work for me when I tried it with the original code before we forked. So, a ticket about Sonarcloud not analyzing it is sooooo far down on my priority list.

uweseimet, 2022-11-06:
@akuker Sending of commands (at least INQUIRY) was working in 22.10.01 as we can see in https://github.com/akuker/RASCSI/issues/131. From what I found (see the commit I mentioned above) something related to the bus changes causes an issue now (INQUIRY not working anymore).
I'm not sure how to proceed here. I mean, your changes are needed to get the Pi working, aren't they? But I remember that you mentioned that these changes could also be merged later. In case it does not cause you a problem to revert the changes (at least for some time) it would help me trying to hopefully make rasdump fully working in the meantime, see ticket https://github.com/akuker/RASCSI/issues/970. rasdump is quite good as a reference when it is about questions like the one in https://github.com/akuker/RASCSI/issues/131.
But again, you should decide. In case we do not temporarily revert the GPIOBUS changes I would still like my changes to rasdump to be merged, because they simplify the code a lot.

uweseimet, 2022-11-06:
@akuker By the way, regarding balance between other projects, life and RaSCSI I know exactly what you mean. I have spent that much time on RaSCSI recently because my list of things that I think should be done/implemented/improved is getting smaller and smaller, and there is not that much left I plan to work on. So it's a lot about finishting things up on my side.

uweseimet, 2022-11-06:
@akuker An update regarding temporarily reverting the bus changes: There was not that much to do except replacing everything in the hal folder by what was in that folder in the v22.10.01 release. The rest of the code (rascsi.cpp in particular) needed hardly any changes, because most changes were backwards compatible. I verified that rascsi is functional again, and all tests are passing. rasdump successfully sends INQUIRY again. Re-adding the new code in hal should be just as straightforward.
From that perspective I suggest to consider temporarily reverting the bus changes. This would pave the way to resolve https://github.com/akuker/RASCSI/issues/969 and https://github.com/akuker/RASCSI/issues/967, and to start with https://github.com/akuker/RASCSI/issues/970. It should not take long to see whether rasdump can be fixed.
Waiting for your feedback.

uweseimet, 2022-11-06:
:-)
```
>sudo  ./bin/fullspec/rasdump -i 0 -b 7 -f dump.hds
RaSCSI hard disk dump utility version 22.11 --DEVELOPMENT BUILD-- (Nov  6 2022, 22:10:52)
Target ID: 0
RaSCSI ID: 7
Vendor:    QUANTUM
Product:   TRB850S
Revision:  0403
Number of sectors: 1660299
Sector size:       512 bytes
Capacity:          810 MiB, 850073088 bytes
Dump progress:   1%(  26752/1660299)
```

closed 2022-11-09 by uweseimet