PiSCSI/piscsi #972: Update Makefile to use clang++

akuker, 2022-11-07:
easyinstall.sh has been updated in PR #971 to install clang++. The Makefile should be updated to use clang by default instead of gcc.

Note: I don't think there is any need to go back to gcc, so I *think* it can be a one-way change.

Ideally, the makefile will still allow for cross-compilation on a x86 system for the ARM platform.

uweseimet, 2022-11-07:
Please see my comment regarding code metrics in https://github.com/akuker/RASCSI/issues/968. Before updating the Makeifile pleease ensure that everything else is still functional. It would be bad to have no code coverage metrics and/or code analysis anymore.

uweseimet, 2023-09-28:
Is this ticket still needed?

closed 2023-10-02 by uweseimet