PiSCSI/piscsi #975: Remove FixCpu, it does not improve the performance

uweseimet, 2022-11-09:
The FixCpu() method should be removed. On a Pi with only one core it is useless, and on a Pi 4 I could not measure any impact on performance/transfer rate.

akuker, 2022-11-10:
What kind of host were you testing with when you confirmed there was no impact on performance? 

The FixCPU() function will only impact the main SCSI/GPIO task, which is the most time critical. Tying this to a single CPU *should* improve performance, since the data/code for this task should remain in that core's cache. I'd like to do more testing before we make this change. 

Is there a pressing need to delete this code?

uweseimet, 2022-11-10:
@akuker More testing is fine, there is no pressing need. If it turns out, though, that there is also no performance improvement on a single CPUI system, this code should be removed.
I will test soon with my Pi Zero.

uweseimet, 2023-03-16:
I would like to finalize this ticket.
I re-tested the transfer rate on a Pi 4. With FixCpu it was 1130 KB/s on average. Without FixCpu it was 1135 KB/s on average. This confirms my previous results: FixCpu does not result in any benefit on a Pi with four cores, it might even result in slightly slower transfer rates. On a single-core Pi it is useless anyway, i.e. testing on a Pi Zero, as I suggested in my last comment, does not make sense.
Based on these results FixCpu should be removed. It just adds useless code and in addition introduces a dependency on Linux.

uweseimet, 2023-06-17:
I added branch issue_975 for finalizing this ticket.

closed 2023-11-24 by uweseimet