PiSCSI/piscsi #977: Move os_integration folder one level up (it is not related to cpp)

uweseimet, 2022-11-10:
The os_integration folder should be moved one level up. Currently it is in the cpp folder, but is not related to cpp.

uweseimet, 2022-11-10:
@rdmark The data in this folder are used by easyinstall, aren't they?

rdmark, 2022-11-11:
@uweseimet One of the files is, namely ```rascsi_bridge```. The easyinstall script copies it into ```/etc/network/interfaces.d/```. I wonder if there's an argument to be made for the Makefile script to do this instead?

uweseimet, 2022-11-11:
@rdmark An additional Makefile target, or adding it to an appropriate existing target, should be fine.

By the way, my old SonarCloud long living feature branches still exists. Can you please remove them? I cannot do htat myself.

rdmark, 2022-11-11:
@uweseimet I cleaned up the Sonar branch reports now. Also, I removed the filter that captured "feature_*" as long lived branches. To your point, feature branches are typically not long lived in this project.

rdmark, 2022-11-11:
BTW, I think we should either remove ```cpp/os_integration/dhcpcd.conf.patch``` from the repo, or have the Makefile build target apply the patch in addition to copying rascsi_bridge. This is the one other part of the wired network bridge setup that easyinstall is currently carrying out. It is a disruptive change since it causes the eth0 IP address to change after a reboot.

uweseimet, 2022-11-11:
@rdmark I don't know what's best here, but the Makefile should not contain any logic that is not directly related to the build/install process.

uweseimet, 2022-11-11:
@rdmark One more remark: It might make sense to have more than one Makefile. A top-level Makefile and, where required) Makefiles in sub-folders. This is a common scenario. If you run make on top-level you build/install everything, if you run make in a sub-folder you only build/install a part.

akuker, 2022-11-11:
Could we please put a hold on moving/renaming things until after I get the banana pi feature merged in? I realize that changes like this make things "better", but they're making it very difficult to keep my branches up to date. I'm spending a non-trivial amount of my time trying to keep my bpi branch up to date with non-functional changes. Its really hard to stay up to date when 50+ files change every few days.

Could we please focus on features/tickets that provide an immediate customer benefit for now?

akuker, 2022-11-11:
As a side note, I'd be glad to get some help with the Banana Pi functionality :)

rdmark, 2022-11-12:
I agree that there's no rush in making changes here. It's just cleanup for a more logical dir tree structure. We could revisit this as a fast follow after the next release.

rdmark, 2023-01-28:
Since there's a lull in activity now, let's address the original request: https://github.com/PiSCSI/piscsi/pull/1085

closed 2023-01-28 by rdmark