PiSCSI/piscsi #979: SCCD has lost the "removable" property

rdmark, 2022-11-11:
In the latest develop, a recent breakage: The SCCD no longer has the ```removable``` property, while SCMO and SCRM retains it.

```
 $ rasctl -T
Supported device types and their properties:
  SCHD  Properties: protectable, stoppable
        Image file support
        Configurable block sizes in bytes: 512, 1024, 2048, 4096
  SCRM  Properties: protectable, stoppable, removable, lockable
        Image file support
        Configurable block sizes in bytes: 512, 1024, 2048, 4096
  SCMO  Properties: protectable, stoppable, removable, lockable
        Image file support
        Configurable block sizes in bytes: 512, 1024, 2048, 4096
  SCCD  Properties: stoppable
        Image file support
        Configurable block sizes in bytes: 512, 2048
  SCBR
  SCDP
  SCHS
  SCLP
```

uweseimet, 2022-11-11:
Fixed in the feature-log-filter branch. Other parameters were missing as well (only in the meta data, not in the actual devices, that why the unit tests did not fail):
```
>rasctl -T
Supported device types and their properties:
  SCHD  Properties: protectable, stoppable
        Image file support
        Configurable block sizes in bytes: 512, 1024, 2048, 4096
  SCRM  Properties: protectable, stoppable, removable, lockable
        Image file support
        Configurable block sizes in bytes: 512, 1024, 2048, 4096
  SCMO  Properties: protectable, stoppable, removable, lockable
        Image file support
        Configurable block sizes in bytes: 512, 1024, 2048, 4096
  SCCD  Properties: read-only, stoppable, removable, lockable
        Image file support
        Configurable block sizes in bytes: 512, 2048
  SCBR  Parameter support
        Default parameters: inet=10.10.20.1/24
                            interface=eth0
  SCDP  Parameter support
        Default parameters: inet=10.10.20.1/24
                            interface=eth0
  SCHS
  SCLP  Parameter support
        Default parameters: cmd=lp -oraw %f
```

rdmark, 2022-11-11:
@uweseimet Verified that it works again now with your branch. Thanks for the quick fix.

PR #978 (2022-11-11): Added rascsi filter to limit logging to a particular device
Also added more unit tests, resolved SonarQube issues, improved code sharing, improved the error handling of the parser and fixed #979.
closed 2022-11-11 by uweseimet