PiSCSI/piscsi #982: Network device interface/interfaces naming discrepancy

rdmark, 2022-11-11:
In the rascsi code, for network type devices we have a property called ```interface``` which takes the comma separated list of network interfaces. However, in the rascsi manpage, the same property is referred to as ```interfaces``` (plural). If you try to use the latter it gets ignored by rascsi.

I have a vague memory that it used to be called ```interfaces``` in the code in the past, but I haven't gone back and confirmed this hunch.

IMHO, the plural form is the more accurate since it can accept a list of multiple interfaces.

uweseimet, 2022-11-12:
@rdmark This was changed to "interface" in 22.05.01 (we talked about it I think, because I would not have changed the networking stuff without you knowing about it), because there was an incompatible change in the semantics related to the introduction of the new "inet" parameter. I'm afraid I forget the exact nature of this semantics change but there was a reason why a new name was needed. "interfaces" was deprecated at that time, and rasctl since 22.05.01 says
```
  SCBR  Supported LUN numbers: 0-31
        Parameter support
        Default parameters: inet=10.10.20.1/24
                            interface=eth0
  SCDP  Supported LUN numbers: 0-31
        Parameter support
        Default parameters: inet=10.10.20.1/24
                            interface=eth0
```
Since 22.05.01 there was a deprecation warning and a TODO related to the removal of the deprecated code:
```
        if (params.count("interfaces")) {
                LOGWARN("You are using the deprecated 'interfaces' parameter. "
                                "Provide the interface list and the IP address/netmask with the 'interface' and 'inet' parameters")

                // TODO Remove the deprecated syntax in a future version
```

uweseimet, 2022-11-12:
@rdmark Just found the related ticket: https://github.com/akuker/RASCSI/pull/684.

rdmark, 2022-11-12:
@uweseimet It's all coming back to me now! Thanks for digging up that old ticket.

So I think the right course of action here is to update the man pages. Let me put up a simple PR for this.

uweseimet, 2022-11-12:
@rdmark OK. I was not aware that the manpages are not up to date.

closed 2022-11-12 by rdmark