PiSCSI/piscsi #985: Images can't be mounted as CD-ROMs

lukexi, 2022-11-13:
# Info

- Which version of Pi are you using: 3B+
- Which github revision of software: develop
- Which board version: 
- Which computer is the RaSCSI connected to: Ensoniq ASR-10

# Describe the issue
I can load disc images if I first rename them to `.hds` to have RaSCSI present them as hard disks instead of CD-ROMs — this works perfectly for Ensoniq-formatted CDs (after my long journey to find my cabling issue), but does not work for e.g. AKAI and Roland-formatted CD-ROM images, which as I understand it the ASR should be able to load — I assume because the ASR only attempts to do translation when it sees a SCSI CD-ROM drive.

Is there pre-existing knowledge of why RaSCSI has limitations when presenting images as a CD-ROM drive? (I see similar workarounds noted for SPARCstations and AKAI samplers). Would love to know before digging in to see if I can figure this one out.

rdmark, 2022-11-13:
My understanding of Akai in particular, is that it's expecting CD-ROM drive volumes with a 512 byte block size, rather than the ISO standard 2048 bytes. For this purpose I've added the "Generic CD-ROM block size 512" drive profile to our database. Can you please try to use this and see if it resolves your problem?

If you're running the latest software, the Web Interface provides a handy way to do this. First you go to the Attach Peripheral Device section, in the CD/DVD row choose to Masquerade As "Generic CD-ROM block size 512", and attach the device. Then in the Current RaSCSI Configuration section, find the device that you just attached, look in the Parameters column, and pick the ISO image to insert from the drop down. That should be it! Now you have an ISO image inserted in an emulated CD-ROM drive that uses the non-standard 512 byte block size.

Please let us know if this makes any difference for your use cases.

lukexi, 2022-11-13:
Thanks Daniel! I just got finished rebuilding again and the new abilities are super cool — the ASR-10 was still unable to read my test Roland and Akai CD images though (failing pretty quickly with `File operation error`). I'll gather some logs in a bit and see if anything shows up!

uweseimet, 2023-10-16:
Any news on this? Without any feedback keeping this ticket permanently open is probably not a good idea.

uweseimet, 2023-11-05:
Closing this ticket because without feedback it cannot be worked on.

closed 2023-11-05 by uweseimet