PiSCSI/piscsi #986: Rename RaSCSI to TBD

akuker, 2022-11-14:
We have been asked by the original developer of RaSCSI to rename this project to something else. The reasoning being that one of the most valuable features to the Japanese Sharp X68000 community was SASI support. Since this project no longer supports SASI, it is not really true to the original "RaSCSI".

Discussion is welcome for what the new name should be.

akuker, 2022-11-14:
I believe we have reached an agreement to continue using the RaSCSI name. Long term, it might make sense to change though.

rdmark, 2022-11-17:
@uweseimet Do you have any ongoing dev branches that would be disrupted by executing on this rebranding? We're trying to nail down the right timing since the codebase wide search and replace will make other dev work difficult for a short while.

rdmark, 2022-11-17:
Thanks for confirming.

uweseimet, 2022-11-18:
@rdmark @akuker Can you outline the scope of the re-branding as far as the C++ code is concerned? In theory one could go as far as renaming files, folders and executables, but I doubt that this is a good idea or required, at least not initially.

There pieces of code with a user-visible reference to RaSCSI:

- The default vendor (INQUIRY) is "RASCSI".
- One of the shutdown modes is called "rascsi".

I don't think there are more code locations like that.

akuker, 2022-11-19:
On hold for now.... the community was not happy about the new name...

akuker, 2022-11-22:
So, the new, new "official" name is PiSCSI. I think we can start planning this now.

Proposed executable changes:
- rascsi -> piscsi
- rasctl -> scsictl
- scsimon -> no change
- rasdump -> scsidump
- scsiloop -> no change (Note: not merged into develop yet)

We should have easyinstall move the old files to a temp location.

Do we want to have symbolic links so that `rascsi` points to `piscsi`? Probably not necessary for people using the web interface.

uweseimet, 2022-11-22:
I don't think that symbolic links are needed. Better to get used to the new names soon ;-).

rdmark, 2022-12-04:
A first stab at this... https://github.com/akuker/RASCSI/pull/1016
Testing in progress.

rdmark, 2022-12-04:
@akuker Are you able to rename the Sonar project key, or will you have to create a new one from scratch?

uweseimet, 2022-12-04:
@rdmark Note that the remote interface and the magic string used in the socket communcation must not be renamed. Otherwise the interface contract is broken at various places.
You can verify that it is *not* broken by running the old rasctl against the new piscsi. Also the shutdown mode "rascsi" must still be supported at least for some time in order to stay backwarods compatible. "piscsi" can be added though. even though one could just stick to rascsi only here.

akuker, 2022-12-04:
I agree. We don't need to completely purge "RaSCSI".

In fact, we should probably outline in the documentation (manpage?) that PiSCSI can not be used at the same time as RaSCSI

akuker, 2022-12-04:
> @akuker Are you able to rename the Sonar project key, or will you have to create a new one from scratch?

Done! I renamed the project key, so it should match your PR

uweseimet, 2022-12-04:
@akuker Nobody reads manpages ;-). IMO the message currently display when starting piscsi twice is sufficient, but it should probably say
```
Error: Port 6868 is in use, is rascsi or piscsi already running?
```
instead of
```
Error: Port 6868 is in use, is piscsi already running?
```

akuker, 2022-12-04:
I like it :)

akuker, 2022-12-04:
I just did a little testing, and the "transfer ownership" function looks to be seamless. Github will automatically re-direct to the new repo name. So, I'm going to take the plunge and re-name/move the repo. Hopefully this doesn't blow up! :)

rdmark, 2022-12-04:
@uweseimet I agree with maintaining backwards compatibility / interoperability for the foreseeable future. It would be advisable to leave code comments in these cases to communicate to future code readers/contributors why certain internal names are different.

Please go ahead and push your changes to the rebrand branch. I'm done rebasing against develop now.

uweseimet, 2022-12-04:
@rdmark Done and tested.

I also fixed one or two SonarCloud issues ... I wonder why new code is still not C++ but rather C-like. Using C++ would reduce the number of SonarCloud issues and would avoid other potential problems.

uweseimet, 2022-12-04:
Can I please get access to the new SonarCloud project? Currently I seem to lack permissions to see it.

rdmark, 2022-12-05:
> Can I please get access to the new SonarCloud project? Currently I seem to lack permissions to see it.

@uweseimet That's odd, the project is Public so anyone should at least be able to see it. How about now? https://sonarcloud.io/project/overview?id=akuker_PISCSI

rdmark, 2022-12-05:
> Note that I did not change the "is ... already running" messages mentioned above.

I pushed a change for this. It makes for a more helpful message potentially.

This should be 99% ready to go now, but I found one instance where the unit test code still referenced "RasUtil". I have a local change and am running the unit tests locally to test before pushing.

uweseimet, 2022-12-05:
@rdmark Your SonarCloud link works, thank you. I was using the search functionality of SonarCloud before, but it did not find the project.

rdmark, 2022-12-05:
The rebrand PR has been merged!

Now, the next task is to add logic to installation scripts to clean up and rename anything "rascsi" before installing, since 1) the two cannot coexist and 2) users expect to have continuity with their user data through the rebranding.

It should be easy enough to add some checks to easyinstall, but I wonder if want the Makefile to check for `/etc/systemd/system/rascsi.service` before installing `piscsi.service` next to it?

rdmark, 2022-12-07:
Data migration introduced to the easyinstall script.

@akuker I believe the only thing left to do is to introduce the new logo, and perhaps finish the wiki editing. Anything else?

rdmark, 2022-12-08:
Made a new ticket for the logo so that we can close this one out https://github.com/PiSCSI/piscsi/issues/1024

closed 2022-12-08 by rdmark