PiSCSI/piscsi #994: Implement support for Banana Pi M2+

akuker, 2022-11-18:
# Info

Because of the supply chain issues, the Raspberry Pi has become nearly impossible to buy at a reasonable price.

# Describe the issue

Request that RaSCSI be updated to work on another SBC platform. Banana Pi M2+ will be the first target platform. Future SBCs may follow in the future.

akuker, 2022-11-18:
Info on Banana Pi M2+
https://wiki.banana-pi.org/Banana_Pi_BPI-M2%2B

bog-dan-ro, 2022-11-24:
I have a bananapi 0 around which, in principle, should have the same pins layout as a rpi0.

akuker, 2022-11-25:
@bog-dan-ro - it will likely require some software updates in order to make it work. The GPIO mapping will probably be different from the BPI M2+. 

This file will likely need to change: https://github.com/akuker/RASCSI/blob/feature_bpi5/cpp/hal/pi_defs/bpi-m2p.h

bog-dan-ro, 2022-11-28:
@akuker thanks for the tips!

At first look it seems you're not using the physical pins instead you're using the logical ones. IMHO using the the physical pins in software will make the code cleaner and maybe the port easier.

I also see you're not using any gpio library (e.g. PIGPIO, WiringPI, etc.) instead you've created your own which probably will make the port easier.

 I'll have a closer look once I get back from my vacation. 
Until then I need to order one of these boards. Which hw version shall I use [2.3](https://github.com/akuker/RASCSI/tree/master/hw/rascsi_2p3) or [2.5](https://github.com/akuker/RASCSI/tree/master/hw/rascsi_2p5) ?

2.3 seems to be available here https://amigakit.amiga.store/rascsi-adapter-board-p-91280.html but I don\t know if it's the same hw as yours or it's changed for Amiga needs.

akuker, 2022-11-28:
Either the 2.3 or the 2.5 will work for you. There's a comparison between the boards here:
https://github.com/akuker/RASCSI/wiki/Hardware-Versions

Version 2.5 will have a brighter external LED (if you're using one). But either one is compatible.

The original RaSCSI code didn't use a gpio library, I'm assuming for performance reasons. So, I just went with that for the BPi as well.

bog-dan-ro, 2022-11-28:
Okay, I'll order a 2.3 premade one and I'll order PCBs if #496 will need additional hw changes.

akuker, 2022-12-04:
This functionality is implemented in develop now. Its slow as molasses, but it there

closed 2022-12-04 by akuker