PiSCSI/piscsi #1001: OLED: Network status does not refresh

nucleogenic, 2022-11-20:
# Info

- Which version of Pi are you using: Raspberry Pi Zero 2 W
- Which github revision of software: develop
- Which board version: N/A
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

If the host hasn't obtained an IP address before the OLED service is launched (e.g. not plugged in to ethernet or connected to WiFi), it will display the no IP address warning until the service is restarted.

The network status should be checked periodically for updates while the process is running.

rdmark, 2022-11-20:
Check this one out! https://github.com/akuker/RASCSI/pull/1002

rdmark, 2022-11-20:
@nucleogenic BTW while testing this I noticed one shortcoming with the current solution: It seems the active network interface's IP address is cached somehow, so if you f.e. unplug the ethernet cable during operation the system still reports the previous IP address. I tried experimenting with `hostname -I` as an alternative way to python sockets to get the IP address, but AFAICT this leads to the same behavior.

Not a huge issue but it would be nice for the script to be aware of a lost network connection.

closed 2022-12-04 by rdmark