PiSCSI/piscsi #1013: Use GNU autoconf for build configuration

uweseimet, 2022-12-01:
While testing the BananaPi changes I noticed that on a Raspberry Pi files that are specific to the BananaPi platform are compiled. Most likely these files are effectively empty (using #ifdefs) but are compiled because they are used by the Makefile.
By using GNU autoconf it should be possible to create a platform-specific Makefile. Benefits:

- The make process only compiles files relevant for the target platform (Banana/Raspberry Pi)
- Shorter compilation times as a result
- The GNU compiler version check in the Makefile can removed. Currently it works more or less by accident, because it checks a g++ version number and does not consider that clang++ version numbers are different.

There might be other benefits.

akuker, 2022-12-01:
I'm in favor of coming up with a different build system/approach. 

But, I don't agree with segregating Banana/Raspberry pi functionality into different executables. IMHO, the code isn't that big for the differences, especially when/if we add different variants of Banana Pi. One of the more requested thing from users is to have pre-built executables, since it does take a long time to compile. I'd like to ultimately combine the different board types (Standard/Fullspec/Aibom/Gamernium) into a single executable as well.

Now that we're "really" using C++ (one could argue the original code was more like C than C++), I think we should also look at CMake. I think it could also provide some of these benefits.

uweseimet, 2022-12-01:
I agree that cmake is an alternative. The current build mechanism may not be sufficient anymore, or at least not in the future, that's the main point I want to make.
Regarding pre-built binaries IMO the most user-friendly approach is a package solution as proposed in (https://github.com/akuker/RASCSI/issues/740.

closed 2023-10-16 by uweseimet