PiSCSI/piscsi #1019: Banana Pi M2+ Performance Fixes

akuker, 2022-12-04:
# Info

- Which version of Pi are you using: Banana Pi M2+
- Which github revision of software: latest develop - Dec 4 2022
- Which board version: All
- Which computer is the RaSCSI connected to: PowerBook G3 Wallstreet

# Describe the issue
When connected to a Banana Pi M2+, the PiSCSI functionality is unusably slow. It seems to function, but is about 5-10% of the expected speed.

uweseimet, 2023-02-27:
@akuker Are there any news on the Banana Pi port? With the current develop branch my Atari does not boot, and the speed is still too low to be usable, just as this ticket says.
Am I the only one who has issues besides the speed? Is there any regular user with a Banana Pi at all?

akuker, 2023-03-02:
No update at this time. You might be the only person currently using BPi.

Due to the way the registers are laid out on the BPi, its likely to be ~3x slower than Raspberry Pi. The GPIO pins are scattered among 3 GPIO registers, where on the Raspberry Pi, they're all in one. The register read/write operations seem to be the bottleneck. 

It may be possible to make a new version of the board (fullspec/gamernium/etc) that is more optimized for the Banana Pi, but i haven't gone down that road yet.

uweseimet, 2023-03-02:
That's a pity, but it confirms the rather bad impression I had of the Banana Pi platform. I guess in a few weeks Raspberry Pis are available again and the Banana Pi is obsolete anyway as far as PiSCSI is concerned ;-).
Besides the performance problem there must be at least another software issue, because my Atari does not boot from the Banana Pi. With the Raspberry Pi there are no problems at all.

akuker, 2023-10-14:
Raspberry Pi supply chain issues seem to be clearing up. The effort to implement this ticket might not be worth it anymore. The Banana Pi functionality could be disabled/removed and this ticket can be closed.

uweseimet, 2023-10-15:
Replaced by #1248.

PR #1022 (2022-12-07): Optimize setting the data gpio bits
This change significantly cuts down on the number of accesses to the GPIO control registers. Testing shows around 800% (8x) speed improvement with this change.

Note that this does not completely resolve #1019. It simply makes it better.

Max read speed is now 166KB/sec
closed 2023-10-15 by uweseimet