PiSCSI/piscsi #1020: Migrate rascsi data in easyinstall.sh

rdmark, 2022-12-06:
This is to break down the requirements for migrating user data and configurations when upgrading rascsi to piscsi.

- [ ] systemd services -- rascsi, rascsi-web, rascsi-oled, rascsi-ctrlboard
- [ ] ~/.config/rascsi
- [ ] ~/.config/piscsi/rascsi_secret (maybe rename to simply call it `secret` -- it's use should be obvious from the context)
- [ ] rascsi-web.key, rascsi-web.crt (maybe we can ignore these, since they get regenerated on the fly?)
- [ ] User group rascsi (this might not need cleanup either)

rdmark, 2022-12-06:
I can see two approaches here: Either have one function that we call every time that checks for the old files all at once and cleans them up, or we can have checks inline whenever we touch the relevant configurations.

rdmark, 2022-12-06:
PR started at https://github.com/PiSCSI/piscsi/pull/1021

akuker, 2022-12-06:
> User group rascsi (this might not need cleanup either)

Long term, it's probably better to rename everything to piscsi unless there is a compelling backwards compatibility concern. 

Or, are you thinking that easyinstall would just use the `rascsi` group name if it already exists? But if we're setting up something new, go with piscsi?

rdmark, 2022-12-06:
@akuker After sleeping on it I agree the group and certs should be cleaned up too. Pushed a commit to the PR. (Still not fully tested though.)

rdmark, 2022-12-06:
Alright, I've tested the PR now. Looks good so far.

closed 2022-12-07 by rdmark