PiSCSI/piscsi #1023: Regression: piscsi version is reported as 0 / 0 / 0 by the protobuf interface

rdmark, 2022-12-07:
The Python clients get 0 / 0 / 0 (major, minor and patch versions) back from the protobuf interface, `message PbVersionInfo` after the piscsi rebranding.

uweseimet, 2022-12-07:
Just for the sake of being more precise: This is not a backend issue but a Python client issue: scsictl and the PiSCSI Control app report the correct version, e.g.
```
>scsictl -s
piscsi server version: 22.12 (development version)
......
```

rdmark, 2022-12-07:
So when I started from scratch with a fresh clone of piscsi on the same system, the issue does not reproduce. So it seems to have to do with some residual data in that "converted" rascsi code.

I tested cloning a fresh 2022 Oct release, and then update to the latest develop. However the issue did *not* reproduce this way.

I would lean towards a Won't Fix for this, since it's happening only in my dirty dev clone.

closed 2022-12-12 by rdmark