PiSCSI/piscsi #1025: Macintosh Plus problems

dor73, 2022-12-08:
# Info

- Which version of Pi are you using: Raspberry Pi Model B Plus Rev 1.2- Raspbian GNU/Linux 11 (bullseye) connected to 

- Which github revision of software: 22.10.01
- Which board version: FULLSPEC Rascsi 2.3b and two termination switch set to "On"
- Which computer is the RaSCSI connected to: Macintosh Plus 1Mb ram

# Describe the issue

i've no 800k floppy working with Macintosh Plus, and i cannot to create it from other old macintosh (i've only a Macbook 2012 or Windows 10 pcs). So i hope to boot my two macintosh plus directly with an image from PiSCSI.

I've tested different prebuilt images from Macintosh Garden or Macintosh Repository, only RaSCSI-Boot-6.0.8.hda from Macintosh Garden image is bootable (https://macintoshgarden.org/apps/rascsi-reloaded). The image RaSCSI-Boot-6.0.8.hda boot without problems with my two Macintosh Plus models (1mb), but when load finder, i cannot then open any programs or control panel or finder menu items. All clic return "Illegal Istructions Error" and the system ask to "Restart". 
Everything happened until yesterday, but today i can't boot anymore the RaSCSI-Boot-6.0.8.hda with boths macs, i've reloaded image from web interface again, but same results.

How can I create a bootable image for Macintosh Plus (1mb) or find a pre built one without using floppy disks?

I've set log to "trace", thanks

marco

Log:

Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.343] [trace] void PrimaryDevice::CheckReady() Device is ready
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.343] [trace] bool DiskTrack::ReadSector(std::vector<unsigned char>&, int) const reading sector: 78
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.343] [trace] void ScsiController::Send() Processing after data collection. Blocks: 4
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.343] [trace] void ScsiController::Send() Continuing to send. Blocks: 4
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.343] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.343] [trace] void ScsiController::Send() Sending handhake with offset 0, length 512
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] bool ScsiController::XferIn(std::vector<unsigned char>&) command=08
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] virtual int Disk::Read(const std::vector<int>&, std::vector<unsigned char>&, uint64_t)
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] void PrimaryDevice::CheckReady() Device is ready
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] bool DiskTrack::ReadSector(std::vector<unsigned char>&, int) const reading sector: 79
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] void ScsiController::Send() Processing after data collection. Blocks: 3
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] void ScsiController::Send() Continuing to send. Blocks: 3
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.347] [trace] void ScsiController::Send() Sending handhake with offset 0, length 512
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.351] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.351] [trace] bool ScsiController::XferIn(std::vector<unsigned char>&) command=08
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.351] [trace] virtual int Disk::Read(const std::vector<int>&, std::vector<unsigned char>&, uint64_t)
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.351] [trace] void PrimaryDevice::CheckReady() Device is ready
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.351] [trace] bool DiskTrack::ReadSector(std::vector<unsigned char>&, int) const reading sector: 80
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.351] [trace] void ScsiController::Send() Processing after data collection. Blocks: 2
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.352] [trace] void ScsiController::Send() Continuing to send. Blocks: 2
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.352] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.352] [trace] void ScsiController::Send() Sending handhake with offset 0, length 512
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.355] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.355] [trace] bool ScsiController::XferIn(std::vector<unsigned char>&) command=08
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.355] [trace] virtual int Disk::Read(const std::vector<int>&, std::vector<unsigned char>&, uint64_t)
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.356] [trace] void PrimaryDevice::CheckReady() Device is ready
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.356] [trace] bool DiskTrack::ReadSector(std::vector<unsigned char>&, int) const reading sector: 81
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.356] [trace] void ScsiController::Send() Processing after data collection. Blocks: 1
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.356] [trace] void ScsiController::Send() Continuing to send. Blocks: 1
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.356] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.356] [trace] void ScsiController::Send() Sending handhake with offset 0, length 512
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] void ScsiController::Send() Move to next phase: datain
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual void ScsiController::Status() Status Phase, status is $00
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] void ScsiController::Send() Move to next phase: status
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual void ScsiController::MsgIn() Message In phase
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 7
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.360] [trace] void ScsiController::Send() Move to next phase: msgin
Dec 08 16:13:49 PISCSI RASCSI[320]: [2022-12-08 16:13:49.361] [trace] virtual void ScsiController::BusFree() Bus free phase

rdmark, 2022-12-09:
@dor73 You can't use an image with the stock Apple hard disk drivers on the Plus. See https://github.com/PiSCSI/piscsi/issues/656

If you look closely at that Mac Garden page you linked, there's a custom MacPack image for use with the Plus, that leverages the Silverlining drivers to work around the incompatibility. Please try that one.

rdmark, 2022-12-12:
Closing for now. Please report back if you test the Silverlining workaround. I'm 99% sure that you are encountering the same issue as described in that other issue.

closed 2022-12-12 by rdmark