PiSCSI/piscsi #1038: RaSCSI on Powerbook 1400 with MacOS 8.6

Jungleboy28, 2022-12-16:
Unfortunately, my Powerbook 1400 no longer recognises my RaSCSI since an update to MacOS 8.6. Mounting hard disks, ISO files or ZIP disks does not work. It also did not boot from the RaSCSI. I have also tried the ready-made images from Macintosh Garden. Nothing is recognised.

I have used SCSIProbe, Lido and all the software available on the RaSCSI bootstrapV3.hda.zip: Nothing worked.

With OS 8.1 everything was still recognised. I was thus able to upload the contents of the RaSCSI-BootstrapV3.hda.zip to the hard disk. After installing on MacOS 8.6, I have these problems.

My RaSCSI has the latest update: October 2022.
PI Zero 2
RaSCSI from Samplerspa (68kmla version)

akuker, 2022-12-23:
HI @Jungleboy28  - Just to make sure I understand properly, everything works OK with OS 8.1. But, with 8.6 you had issues?

What did SCSIProbe report? Did it see the drives at all? Or did it just fail to mount the drives?

Jungleboy28, 2022-12-25:
Yes, with OS 8.1 RaSCSI works okay. 

I used SCSIProbe 4.3 today and RaSCSI v3 Bootrap was now detected. My other hard drives are also suddenly recognised. Last week it did not work. Strange. It is copying my games to the hard disk.

But ISO files are not recognised as CD-ROM. I downloaded the ISO file from Macintosh Garden ADDY Kids German and it is not recognised.

Jungleboy28, 2022-12-25:
I have restarted the computer and now I get the error message in SCSIProbe 4.3: Bus Not Terminated. It recognises a connection but interprets it as NO DATA.

Jungleboy28, 2022-12-25:
Okay, strange behaviour of the RaSCSI. It only recognises the hard disks from RaSCSI when I start the Powerbook 1400 without a SCSI connection connected. It recognises the hard disk when I connect the RaSCSI to the Powerbook 1400 after a reboot. The system must be fully loaded.

Is this normal?

rdmark, 2022-12-29:
>But ISO files are not recognised as CD-ROM. I downloaded the ISO file from Macintosh Garden ADDY Kids German and it is not recognised.

Please note that the stock Apple CD-ROM extension can be picky with which ISO images it will recognize when used in conjunction with PiSCSI. You can try other CD-ROM drivers such as RapidCD. Read more at https://github.com/PiSCSI/piscsi/wiki/Drive-Setup#mac-os-drivers

uweseimet, 2023-10-16:
I'm just browsing stale tickets. Any idea how (or if) to proceed with this ticket? IMO this might also be a termination issue.

uweseimet, 2023-11-05:
Closing this ticket because without feedback it cannot be worked on.

closed 2023-11-05 by uweseimet