PiSCSI/piscsi #1053: piscsi: assertion in localizer.cpp fails

uweseimet, 2022-12-22:
When launching piscsi (develop branch) with LANG=en_US.UTF-8 an assertion fails.

```
piscsi: ./piscsi/localizer.cpp:238: void Localizer::Add(LocalizationKey, const string&, std::string_view): Assertion `supported_languages.find(locale) != supported_languages.end()' failed.
Aborted
```

This must have been caused by a quite recent change.

uweseimet, 2022-12-22:
Not directly related, but there is duplicate/redundant logging, at least on regular PC hardware:

```
[2022-12-22 22:32:46.336] [error] Unknown type of sbc detected: 0
[2022-12-22 22:32:46.336] [info] Detected device Unknown SBC
```

rdmark, 2022-12-22:
@uweseimet Do you want to try to address this bug before we tag the next release? I think @akuker is planning to tag the release tonight his time, or perhaps tomorrow.

uweseimet, 2022-12-23:
@rdmark I´m afraid I am busy with other things. On first sight a new language was added but only in localizer.cpp but not in localizer.h. Whoever committed the latest translations should investigate this before the release, because it is likely that the language handling is broken and others will also stumble upon it, not just me.

rdmark, 2022-12-23:
Good catch. This should fix the missing code. https://github.com/PiSCSI/piscsi/pull/1055

uweseimet, 2022-12-23:
@rdmark I could run a quick test, and the problem appears to be resolved, thank you.

closed 2022-12-23 by uweseimet