PiSCSI/piscsi #1065: [idea] Use SPI of Raspberry Pi with Shift-Registers / SPI to Parallel to improve performance

schreinerman, 2023-01-08:
# Info

[idea] Use SPI of Raspberry Pi with Shift-Registers / SPI to Parallel to improve performance

# Describe the issue

Raspberry Pi can drive up to 250MHz with its SPI. If used with Shift-Registers like 74HC/HCT165 (up to 48MHz), so performance could be increased. Some Shift-Registers can already do the level-shifting, so level-shifters could get obsolete. SPI usage would also improve portability to non-Raspberry Pi HW.

akuker, 2023-01-14:
Very cool idea!! Thank you for sharing!

schreinerman, 2023-01-14:
Eventuelly an SPI port / IO expander could also make sense... I googled a little bit and MAX7301 can drive up to 26MHz. But sure 26MHz has to be divided by the used bits. But there should be also others... maybe with Linux Kernel (module) support or library which could make the implementation more simple.

akuker, 2023-01-23:
Have you seen any documentation confirming that the Raspberry Pi can support 26MHz over its I2C bus? From the data sheet, it looks like the SoC has a limit of 400Kbps

https://datasheets.raspberrypi.com/bcm2711/bcm2711-peripherals.pdf

schreinerman, 2023-01-23:
I2C is limited to 400Kbps as specified, correct. But I was not speaking about I2C. I was speaking about SPI. SPI is also used for attaching RAM, Flash, Displays, etc. which are requiring much higher speeds. That's why it is supporting higher speed transfers on most chips. 

Because it is more simple than I2C and therefore requiring a chip select for each device in the line, data can be converted very simple with shift registers with data in(/out) + clock in.

For maximum speed supported, see also BCM2708 Linux drivers (line 208): https://github.com/bootc/linux/blob/073993b3f3e23fb8d376f9e159eee410968e0c57/arch/arm/mach-bcm2708/bcm2708.c

akuker, 2023-01-23:
You are correct! I got my SPI / I2C wires crossed :-\

The BCM2708 linux driver says the max speed is 500000 Hz (5kHz)
```
	{
		.modalias = "spidev",
		.max_speed_hz = 500000,
		.bus_num = 0,
		.chip_select = 0,
		.mode = SPI_MODE_0,
	},
```

So, if the CLK signal going to shift register is running at 5 kHz, that's one bit every 200us. Assuming we're only using the shift registers for the 8 data lines, that's 1.6ms per byte (if we ignore the data parity and control signals). Rough math, that's around 625 K bytes per second. The current PiSCSI software can do around 1 M bytes per second. 

Wouldn't the shift register be slower? (I could be wrong on my math here!! I'm still recovering from a cold)

guruthree, 2023-04-20:
It seems like the BCM2708 is the die name, while BCM2835 is the part as it's packaged for the Pi [1].

I can confirm I've been able to run shift registers with SPI at speeds of around 25 MHz before, although not specifically with a Pi. It looks like this SPI level should be reachable on the Pi, although not much more than this [2].

[1] https://raspberrypi.stackexchange.com/questions/840/why-is-the-cpu-sometimes-referred-to-as-bcm2708-sometimes-bcm2835
[2] https://elinux.org/index.php?title=RPi_SPI