PiSCSI/piscsi #1068: [fix] Generated heat by drivers and termination resistors

schreinerman, 2023-01-11:
# Info

- Which version of Pi are you using: Raspberry Pi Zero W
- Which github revision of software: Latest
- Which board version: 2.4a
- Which computer is the PiSCSI connected to: PowerBook 170

# Describe the issue

Driver chips are getting hot and termination resistors are drawing 200mA and generating additional heat. 

Driver chips getting colder by adding 100nF decoupling capacitors.

Cutting 5V from TERM_5v, adding an 3.3V LDO (5V input, 3.3V output) and connecting 3.3V to TERM_5v is minimising the generated heat dramatically (!)

akuker, 2023-01-14:
The newer versions of the board also include higher rated resistors (in a 1210 package instead of 0402). This reduces the heat from the termination resistors as well.

akuker, 2023-01-14:
> Cutting 5V from TERM_5v, adding an 3.3V LDO (5V input, 3.3V output) and connecting 3.3V to TERM_5v is minimizing the generated heat dramatically (!)

Doesn't this push the termination voltage out of spec? The termination voltage would be 1.98v instead of 3v. It would definitely reduce heat, but I'm not sure it would work reliably in all cases.

schreinerman, 2023-01-14:
forgot to mention: leaving the GND resistors unconnected (so 0.3V over spec, so better would be to use an adjustable LDO and set it to 3V)

ManuTester, 2023-05-02:
While you're at it, you cannot turn off the termination by just removing the 5v and GND from the resistors, because the resistors are still linked and cause a connection between the signals. If you want jumpers, you need to use one jumper per signal! So the jumpers as they are, are pretty pointless - except to apply the mod schreinerman describes :)

For decades, socketed resistor networks were the de facto standard for passive termination so maybe stick with that? If one does not want to have the termination switchable, it's always possible to solder the resistor networks directly without socket. The resistors networks are pretty cheap as well!

Here is another explanation of the solution schreinerman proposed:
https://www.amigawiki.org/doku.php?id=de:signals:scsi_termination
No current flows from Vcc to GND, so much lower power consumption.

This does not fix the problem of having to remove all resistors to turn off termination, though.