PiSCSI/piscsi #1072: Transport endpoint is not connected, chown: invalid group

nsputnik, 2023-01-20:
# Info

- Which version of Pi are you using: 3B+
- Which github revision of software: latest
- Which board version: 2.4a
- Which computer is the PiSCSI connected to:none yet but will be PM 7200

# Describe the issue

I am stuck on step 5 of the install. After I run each of the commands it say that they failed and "Transport endpoint is not connected." When I try to install Samba it says "creating the /home/rascsi/shared_files directory and granting read/write permissions to all users...
chown: invalid group: ‘rascsi:rascsi’" rascsi is my username on the Pi. What am I doing wrong?

rdmark, 2023-01-20:
What this tells me is that the version of easyinstall.sh that you're running is too old, since it's trying to assign your user to the group `rascsi` (the pre-Dec 2022 behavior).

You may have local changes that is preventing the repo from updating. Try this:

```
git stash
git checkout develop
git pull origin develop --prune
```

...and then run easyinstall again.

nsputnik, 2023-01-21:
When I try `git stash` I get:
fatal: not a git repository (or any of the parent directories): .git

rdmark, 2023-01-21:
Right, you need to run those commands from within a cloned piscsi code repo.

nsputnik, 2023-01-21:
OK, I installed again and I was able to run the lines at step 5 but I am not getting a web interface at the IP of the Pi.  I installed Samab but it will not let me log in with the Pi's username and the password Samba asked me to pick at the install.  The last time I set up a RaSCSI was a few years ago and I did not run into all of these issues.

nsputnik, 2023-01-21:
I still get `usermod: group 'rascsi' does not exist`

rdmark, 2023-01-21:
Can you please provide the complete terminal logs of what led up to these issues. It's impossible to tell exactly what's going on with just one liner errors.

Please make sure that the ip address that you're trying to access the Pi remotely with is the right one: use the command `ip addr show` and make sure you can ping it.

nsputnik, 2023-01-21:
When I rane `git pull origin main --prune` it returned:

```
Pulling without specifying how to reconcile divergent branches is discouraged. You can squelch this message by running one of the following commands sometime before your next pull:

git config pull.rebase false  # merge (the default strategy)
git config pull.rebase true   # rebase
git config pull.ff only       # fast-forward only

You can replace "git config" with "git config --global" to set a default preference for all repositories. You can also pass --rebase, --no-rebase, or --ff-only on the command line to override the configured default per invocation.
```

rdmark, 2023-01-21:
Again, please share the full context of these messages. The above is very common message that you get when you haven't configured the default merging strategy. Just do exactly what it says and pick your preferred strategy. It is likely that it fell back to fast-forward afterwards, which should be obvious if you share the logs.

nsputnik, 2023-01-21:
I'm sure it is the right IP address.  I used it to ssd in and I can see it on the OLED.  Been using RaSCSI for 2 years, just trying to set up another one now with the current version.  It is a fresh install of Bullseye so I am not sure why they permission issues are happening.

nsputnik, 2023-01-21:
```
rascsi@raspberrypi:~/piscsi $ ./easyinstall.sh
    .~~.   .~~.
   '. \ ' ' / .'
    .╔═══════╗.
   : ║|¯¯¯¯¯|║ :
  ~ (║|_____|║) ~
 ( : ║ .  __ ║ : )
  ~ .╚╦═════╦╝. ~
   (  ¯¯¯¯¯¯¯  ) PiSCSI Assistant
    '~ .~~~. ~'
        '~'

Board Type: FULLSPEC | Compiler: clang++-11 | Compiler Cores: 2

Choose among the following options:
INSTALL/UPDATE PISCSI
  1) Install or update PiSCSI Service + Web Interface
  2) Install or update PiSCSI Service
  3) Install or update PiSCSI OLED Screen (requires hardware)
  4) Install or update PiSCSI Control Board UI (requires hardware)
NETWORK BRIDGE ASSISTANT
  5) Configure network bridge for Ethernet (DHCP)
  6) Configure network bridge for WiFi (static IP + NAT)
INSTALL COMPANION APPS
  7) Install AppleShare File Server (Netatalk)
  8) Install FTP File Server (vsftpd)
  9) Install SMB File Server (Samba)
 10) Install Web Proxy Server (Macproxy)
ADVANCED OPTIONS
 11) Compile and install PiSCSI stand-alone
 12) Configure the PiSCSI Web Interface stand-alone
 13) Enable or disable PiSCSI back-end authentication
 14) Enable or disable PiSCSI Web Interface authentication
EXPERIMENTAL FEATURES
 15) Share the images dir over AppleShare (requires Netatalk)
 16) Compile PiSCSI binaries
Enter your choice (0-16) or CTRL-C to exit: 1
Installing / Updating PiSCSI Service (FULLSPEC) + Web Interface
This script will make the following changes to your system:
- Install additional packages with apt-get
- Add and modify systemd services
- Install the Nginx web server
- Create files and directories
- Change permissions of files and directories
- Modify user groups and permissions
- Install binaries to /usr/local/bin
- Install manpages to /usr/local/man
- Create a self-signed certificate in /etc/ssl
Input your password to allow this script to make the above changes.
[sudo] password for rascsi: 
The /home/rascsi/images directory already exists.
The /home/rascsi/.config/piscsi directory already exists.
Updating checked out git branch origin/main
Already up to date.
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
Compiling FULLSPEC with clang++-11 on 2 simultaneous cores...
rm -rf obj/fullspec bin/fullspec generated coverage piscsi.dat
-- Generating protobuf-based source files
-- Creating directory obj/fullspec
mkdir -p generated
mkdir -p obj/fullspec
protoc --cpp_out=generated piscsi_interface.proto
-- Creating directory bin/fullspec
mkdir -p bin/fullspec
clang++-11 -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++17 -iquote . -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -MD -MP  -DFMT_HEADER_ONLY -DCONNECT_TYPE_FULLSPEC -c ./scsictl/scsictl_display.cpp -o obj/fullspec/scsictl_display.o
mv generated/piscsi_interface.pb.cc generated/piscsi_interface.pb.cpp
clang++-11 -O3 -Wall -Werror -Wextra -DNDEBUG -Wno-psabi -std=c++17 -iquote . -D_FILE_OFFSET_BITS=64 -D_LARGEFILE64_SOURCE -MD -MP  -DFMT_HEADER_ONLY -DCONNECT_TYPE_FULLSPEC -c ./scsictl/scsictl_core.cpp -o obj/fullspec/scsictl_core.o
[... 313 lines cut ...]
(samba-ad-dc.service masked)
Created symlink /etc/systemd/system/multi-user.target.wants/nmbd.service → /lib/systemd/system/nmbd.service.
Failed to preset unit: Unit file /etc/systemd/system/samba-ad-dc.service is masked.
/usr/bin/deb-systemd-helper: error: systemctl preset failed on samba-ad-dc.service: No such file or directory
Created symlink /etc/systemd/system/multi-user.target.wants/smbd.service → /lib/systemd/system/smbd.service.
samba-ad-dc.service is a disabled or a static unit, not starting it.
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u4) ...

Modifying /etc/samba/smb.conf ...
server min prototol = NT1

[Pi File Server]
path = /home/rascsi/shared_files
browseable = yes
writeable = yes
hide dot files = yes
veto files = /.*/
Please create a Samba password for user rascsi
New SMB password:
Retype new SMB password:
Added user rascsi.
Installing SMB File Server - Complete!
rascsi@raspberrypi:~/piscsi $ 
```

rdmark, 2023-01-21:
Is it a fresh Bullseye that you installed from scratch, or did you use the pre configured release image? https://github.com/PiSCSI/piscsi/releases/tag/v22.12.01

nsputnik, 2023-01-21:
I did not use the image at https://github.com/PiSCSI/piscsi/releases/tag/v22.12.01 because I could not find it but now I see it at the bottom.  The image I used from from https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2022-09-26/2022-09-22-raspios-bullseye-armhf-lite.img.xz

rdmark, 2023-01-21:
Either method should work. If you use the pre configured image you bypass some of the issues you've been having.

rdmark, 2023-01-21:
The reason the web interface isn't working for you is that the script clearly fails in the `installPiscsiWebInterface()` function. This is the `usermod: group 'rascsi' does not exist` message that you call out.

It is probably this line that fails: `sudo usermod -a -G $USER www-data`. I think $USER and www-data should be the other way around actually.

rdmark, 2023-01-21:
In fact, that usermod line can be removed I think. It's something that was carried over from an earlier permissions scheme.

rdmark, 2023-01-21:
This PR should fix it https://github.com/PiSCSI/piscsi/pull/1073

nsputnik, 2023-01-21:
I have used the image you have provided to start over.  Everything is good, I figured out I did not need to install it, which maybe the docs can explain.  I installed Samba from the script but now I can't write anything to the directores when logged in from my MacOS.  How can I make these directory writable, and why can't they be writable by default?  I like to use Samba to shuttle the image files because the web interface (at least the one from a few years ago) would not show any progress markers.

rdmark, 2023-01-21:
I have frankly only used the Samba server with Windows 98SE and XP clients. You may have to create a Samba user that matches the exact macOS user that you want to have write permissions to the file server (both username and password.)

rdmark, 2023-01-21:
The fix for easyinstall has been merged. Please report any other issues that you encounter in separate tickets.

closed 2023-01-21 by rdmark