PiSCSI/piscsi #1077: How to properly PHYSICALLY connect a RaSCSI Reloaded board to a Pi to Macintosh II?

MarcTremblay1981, 2023-01-27:
Hi, first time poster and PiSCSI user here!

- Which version of Pi are you using: Pi Zero W v1.1
- Which github revision of software: PiSCSI 22.12.1
- Which board version: RaSCSI Reloaded Zero v1.0 FULLSPEC
- Which computer is the PiSCSI connected to: Macintosh II (but not yet)

I bought 2 new, pre-assembled RaSCSI Reloaded boards from a seller on eBay and he did not include ANY documentation or install instructions.  I found this GitHub and followed the compile instructions and successfully installed PiSCSI on the Pi Zero W and everything SEEMINGLY works.  I can connect to the PiSCSI Control Page in my web browser and upload, expand and attach devices to the virtual SCSI chain.  All the software side of it, I got it working without any issue.

What bothers me is I don't understand how to properly attach it to the Pi, and then the assembly to the Macintosh II to actually make it work.  I had a couple basic questions:

A) Do I have to connect an USB power lead to the RaSCSI Reloaded board? (see Ref-1)
B) Do I have to connect an USB power lead to the Pi Zero W? (see Ref-2)
C) Does the SCSI port on the RaSCSI need to be plugged in the Mac for the Pi to turn on? (see Ref-3)
D) What is the proper orientation of the GPIO hat? (see Ref-4 and Ref-5)
E) Are both jumper pads supposed to be soldered on the back of the RaSCSI? (see Ref-6)

- When I power on the Pi *WITHOUT* the RaSCSI plugged in at all, everything works, the Pi boots up and the PiSCSI Control Page shows up in my web browser after a minute.
- When I power on only the Pi *WITH* the RaSCSI plugged in the Pi, then the Pi does not turn on, no LED at all on any of the two boards.
- When I power on only the RaSCSI  *WITH* Pi plugged in, then the 5V LED is solid on the RaSCSI but the Pi does not turn on.
- When I power both boards, the 5V LED is solid on the RaSCSI but the Pi does not turn on and it makes a high pitch buzzing sound.

Thank you so much to anyone who can help me figure out how to properly connect this device.  I don't understand how there seemingly is no connection diagram anywhere for this device.  I think this would really help beginners like me at least connect the PiSCSI device properly and get started.

[image]
[image]
[image]
[image]

akuker, 2023-01-27:
> A) Do I have to connect an USB power lead to the RaSCSI Reloaded board? (see Ref-1)

You can connect the power to EITHER the PiSCSI (RaSCSI Reloaded)  OR the Raspberry Pi Zero. The port on the PiSCSI is intended to just be for convenience. When you assemble the PiSCSI Zero with the Pi on top, the SCSI connector will block the power connector. 

> B) Do I have to connect an USB power lead to the Pi Zero W? (see Ref-2)

Either port works. But NOT BOTH

> C) Does the SCSI port on the RaSCSI need to be plugged in the Mac for the Pi to turn on? (see Ref-3)

No. The PiSCSI should function stand-alone when power is applied

> D) What is the proper orientation of the GPIO hat? (see Ref-4 and Ref-5) Note: According to the GND silk on the RaSCSI, it should be as shown on Ref-4, please correct me if I'm mistaken)

The PiSCSI/RaSCSI you received is in the "Alternate" configuration on this page: https://github.com/PiSCSI/piscsi/wiki/PiSCSI-Zero-Assembly

[image]

> E) Are both jumper pads supposed to be soldered on the back of the RaSCSI? (see Ref-6)

It depends on how you're using your PiSCSI. If you're using the PiSCSI Zero, 99.9% of the time you want the term enb pads to be bridged (enabling termination).  Your picture # 6 matches having termination enabled.

The general goal is that the last device on each end of the scsi bus should be terminated. With the PiSCSI Zero, there is only one connector, so it has to be at the end of the chain.

Page 49 of this doc has a little more information: https://vintageapple.org/macmanuals/pdf/Power_Macintosh_Users_Guide_1995.pdf
[image]

With the jumper bridged, its enables termination. With the jumpers cut (disconnected), termination is disabled

I hope this helps!!

MarcTremblay1981, 2023-01-27:
I literally just discovered one very, very important detail that I could not find anywhere except by taking a much more attentive look to the seller's pictures: It turns out that the GPIO header has to be soldered TO THE BACK of the Pi, not to the front (standard).  This sounds like a very basic thing to consider, but I did not pay attention to the pictures enough and just assumed that the GPIO header was standard when it's in fact backwards.

akuker, 2023-01-27:
You are correct!

rdmark, 2023-01-29:
Since this issue seems to be incorrectly assembled hardware, let us close it for now.

closed 2023-01-29 by rdmark