PiSCSI/piscsi #1081: piscsi doesn't gracefully handle bad (I think?) images

vvuk, 2023-01-28:
# Info

- Which version of Pi are you using: Pi Zero W
- Which github revision of software:  Dec 2022
- Which board version: 4.4
- Which computer is the PiSCSI connected to: Sparcstation 2

I've got an `.iso` that I don't think is actually a valid ISO -- there's some header junk that doesn't look right.  Trying to mount it via the web UI results in an Internal Server Error.  Doing it on the command line results in `Error: Invalid protobuf message header`.

Would be nice if something nicer was reported here.  (Still TBD whether it's a bad image or not, but some error would help me figure that out too :) )

rdmark, 2023-01-28:
@vvuk Can you please share the iso image in question with us?

vvuk, 2023-01-28:
Yep -- should be this one: https://fsck.technology/software/Sun%20Microsystems/SunOS%20Install%20Media/SunOS%204.1.4%20SPARC%20%28CD%29/sunos_4.1.4_install.iso

rdmark, 2023-01-28:
Thanks! This is the actual error message:

```
$ sudo piscsi -ID0 sunos_4.1.4_install.iso
SCSI Target Emulator PiSCSI (Backend Service)
Version 22.12 --DEVELOPMENT BUILD--  (Dec  7 2022 11:36:33)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2022 Contributors to the PiSCSI project
Connection type: FULLSPEC
[2023-01-27 19:23:42.093] [info] Set log level to 'info'
[2023-01-27 19:23:42.093] [info] Detected device Raspberry Pi 2/3
[2023-01-27 19:23:42.104] [info] Validating: operation=ATTACH, device id=0, lun=0, type=UNDEFINED, device params='file=sunos_4.1.4_install.iso', vendor='', product='', revision='', block size=0
terminate called after throwing an instance of 'io_exception'
  what():  Raw ISO CD-ROM file size must be a multiple of 2536 bytes but is 378178080 bytes
Aborted
```

uweseimet, 2023-01-28:
The image size is wrong, just as the error message says. But I think we can make this a warning, see https://github.com/PiSCSI/piscsi/pull/1084.

vvuk, 2023-01-28:
Right, the issue is that error message doesn't show up in the web ui, or via scsictl. It doesn't get passed through the protocol. In both of those you just get a generic hard error.

uweseimet, 2023-01-28:
The error message is not propagated to the client (e.g. web UI,  app or scsictl) because the piscsi process terminates. With the fix_issue_1081 feature branch and the related PR this should be resolved. There is only a warning now, because the image does not necesarily have to be rejected, because except for the excess bytes it may contain valid data. That's the same behavior we already have for hard disk images which are not multiples of the physical sector size: The excess bytes are ignored.

uweseimet, 2023-01-28:
@vvuk The fix has been merged into the develop branch. Can you please test whether it works for you?

rdmark, 2023-01-28:
I can confirm that piscsi doesn't abort anymore, and throws a warning log message instead.

uweseimet, 2023-01-28:
@rdmark Sounds good. I am going to close this ticket, because I do not expect @vvuk to still have an issue.

closed 2023-01-28 by uweseimet

vvuk, 2023-01-28:
Thank you!  I'll confirm when I get a chance -- I'm currently juggling piscsi, scsi2sd, and bluescsi on various computers, need to get the piscsi back in order :)