PiSCSI/piscsi #1096: IP address on control board

levo-man, 2023-02-11:
# Info
Looking for a way to display the IP address of the Pi with the control board software.

- Which version of Pi are you using:
- Which github revision of software:
- Which board version:
- Which computer is the PiSCSI connected to:

# Describe the issue
Looking for a way to display the IP address of the Pi with the control board software. Either as a menu item , or at the bottom of the list of drives ( like the OLED software does ) . 
If as a menu item would it be possible to show the IP for any interface that is connected wlan0 or eth0 but only if it has an IP assigned.

rdmark, 2023-02-24:
@bzeiss Is this something that you had on your roadmap?

bzeiss, 2023-02-24:
It's a good idea. However, before working on new features, I want to check whether the 'break-in' problem of the rotary is something that can be solved. I suspect it's a problem inherent to the fact that the rotary goes through the multiplexer, usage of python for timing sensitive events as a rather slow language and the raspberry pi os doing a lot of other things besides looking for rotary encoder events.

uweseimet, 2023-10-16:
@bzeiss I'm looking for stale tickets and have stumbled upon this one. Any news on this?

levo-man, 2023-10-16:
> From "Uwe Seimet" ***@***.***>
> To "PiSCSI/piscsi" ***@***.***>
> Cc "levo-man" ***@***.***>; "Author"
> Date 16/10/2023 22:56:52
> Subject Re: [PiSCSI/piscsi] IP address on control board (Issue #1096)

I don't know about 'stale' but it is a project that is still available 
for people to build .

uweseimet, 2023-10-17:
@levo-man Sure, but I was wondering whether @bzeiss has an updated status. Nobody else appears to be involved in this ticket.

bzeiss, 2023-10-17:
Yes, fair question. My current impression is that there is little interest in the control board and I'm not sure whether it's worth spending a lot of time investigating a problem that is probably better solved with a different hardware design. Rotarys over I2C multiplexer are probably just not a very good idea at the end of the day (especially if the operating system is doing different things like scsi emulation at the same time). To be honest, we should probably think about removing the control board code.
@akuker @rdmark

levo-man, 2023-10-17:
> From "Benjamin Zeiss" ***@***.***>
> To "PiSCSI/piscsi" ***@***.***>
> Cc "levo-man" ***@***.***>; "Mention"
> Date 17/10/2023 09:13:18

Or , maybe just do a redesign of the control board . I do like the 
functionality of it and will be doing a blog on a build soon .

akuker, 2023-10-18:
I don't think the control board is dead. There haven't been many recent updates, because its relatively simple and works.

A re-design of the controller board would be good. It should have its own microcontroller to handle the rotary encoder so the Pi doesn't have to deal with it.

levo-man, 2023-10-18:
Yes I think tha that might be the way to go , if a rotary is definitely 
needed . The trouble is that I am stacked with things to repair , and 
projects to do at the moment .
Something I would definitely be willing to look at in the future .

alan .

From "Tony Kuker" ***@***.***>

Date 18/10/2023 02:47:14

rdmark, 2026-01-06:
FTR, this is how I designed it. You go to the menu for any SCSI ID, and then scroll to the bottom where you can find a **System Info** menu item.

[image]
[image]